f4exb/sdrangel #732: Icones location airports

choupette-change, 2020-12-09:
Hi everyone, the position of the airports and heliports locations icons seem to be incorrect in the SdrAngel 6.3.0 version of Ubuntu 20.04 (ok for the Windows version) in plugin adsb demod.

choupette-change, 2020-12-09:
Sorry in Ubuntu 20.10

srcejon, 2020-12-09:
Do you have a screen shot showing the problem?

choupette-change, 2020-12-10:
[SDRANGEL.docx](https://github.com/f4exb/sdrangel/files/5672166/SDRANGEL.docx)

srcejon, 2020-12-10:
I've just built the latest source (6.3.0-3-gba642e8fa) on Ubutnu 20.10 and those two airports appear in the correct locations for me.

Have you tried redownloading the airport database?

Did you build SDRangel from source yourself? What options did you build with? What version of gcc?

choupette-change, 2020-12-10:
1 Have you tried redownloading the airport database?
Yes several times (airports & heliports), all the airports that I know in my region are out of step. as stated in my previous post, it's ok with Windows ...
2 Did you build SDRangel from source yourself ?
Yes https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux step by step , I reloaded the prerequisites ,
I use a RTLSDR with Ubuntu 20.10
What version of gcc?
$ gcc --version
gcc (Ubuntu 10.2.0-13ubuntu1) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE
[image]

srcejon, 2020-12-10:
Ok, thanks. Sounds pretty much the same as I've got here, so not sure what the problem might be, I'm afraid.

choupette-change, 2020-12-12:
I have completely uninstalled and reinstalled step by step the latest version of SDRANGEL https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux and Ubuntu 20.10 and I still have the same problem with the icons of airports and heliports, they are not displayed in the right place on the aircraft map . I have positioned: "Airport display distance": 3000 km and I see that the icons are displayed on parallel lines ???? (see screenshot).
For information My position is displayed correctly on the map and the icons are ok on the "vor map". The windows version of SDRANGEL correctly displays all the icons (airport and heliport) correctly. I don't understand, can someone help me ?? Thanks in advance
[image]
[image]

closed 2021-01-24 by f4exb