f4exb/sdrangel #733: "Frequency" and "relative position of device center frequency" are modified after switching from RX to TX and back

RaspCla, 2020-12-11:
Hello,
I'm using version 6.3.0 / Windows 10 Pro, 64bit, version 10.0.19042 (most current verion).
My hardware is a Hackrf one.

If I start receiving e.g. at 145.650 Mhz and are switching to TX (configured to 145.050Mhz) via simple PTT (or manually), the RX "frequency" and "relative position of device center frequency" have been changed from previous values.
RX "frequency" is changed to 145.400 Mhz, "relative position of device center frequency" is changed to "Cen", independent if it was set to "Inf" odr "Sup" before.
(sample rate is 1000 kS/s)

I'm using NFM modulator and demodulator, see picture attached
Hope you can help

[image]

f4exb, 2020-12-13:
Firstly I see you are using SoapySDR with HackRF. Why do you do that? HackRF has native support so you could use the HackRF plugin that by the way shows up in the list so you HackRF is recognized by the plugin.

Secondly you have to realize that HackRF is a half duplex device that has a single LO for Rx and Tx and therefore cannot use different Rx and Tx frequencies thus the last one (Rx or Tx) "wins". This is normal behavior.

RaspCla, 2020-12-13:
If I use the native support (see attachment), I get a error message "Could not start sample source" as soon I start e.g. receiving.
Sometimes even SDRangel is crashing immidiately after selecting HackRF as device.

In regard to half duplex device:
I was aware of this fact but assumed that SDRangel will change the frequency while transmitting and change back after transmitting. 
That means I can not use HackRF e.g. for relais etc. 

Thanks a lot for answering

f4exb, 2020-12-13:
First point is covered with #718 but no luck so far... There is no assumption on a different frequency for Rx and Tx for any device. Some devices have independent Rx and Tx LOs (like LimeSDR) so this is essentially device dependent. However a more advanced version of PTT may implement this. I have this in mind but not elaborated enough yet to put it in a ticket.

RaspCla, 2020-12-13:
Ok, would be great (new version of PTT).
I just hade some additional issues, the "Relative position of device center frequency" is always at center (noise at waterfall), even if I select "Sup" or "Inf". But now I'm able to use 145,050Mhz for TX and 145.650Mhz for RX while using the PTT button.
Even if I switch several times forth and back it works fine.

RaspCla, 2020-12-14:
I did some additional investigations. If I use "Relative position of device center frequency" = "Cen" 
I can switch between both frequencies (RX <-> TX) and the frequency's are always swtiched to the defined values, correctly.
If I set "Relative position of device center frequency" to "Sup" or "Inf" the RX frequency is modified while switching to TX.
Additionally the "Relative position of device center frequency" is switched back to "Cen"
I've the feeling the issue with wrong frequency is because the "Relative position of device center frequency" is not restored before returning from TX to RX. Just an idea.
Is there a possibility to change this behavior?

Additionally I found out, that I can not set a "Relative position of device center frequency" <> "Cen" if the "Software decimation factor" = "1" (it works with "2", "4", "8")

f4exb, 2020-12-15:
The "Relative position of device center frequency" affects how the transferred I/Q baseband from the device to the host is decimated. If decimation is 1 (i.e. no decimation) this setting has of course no effect.

f4exb, 2021-04-28:
Questions answered. Handling frequency shifts in PTT is not planned or may be part of a new ticket.

closed 2021-04-28 by f4exb