f4exb/sdrangel #739: Program will not start after update to 6.3.3 on Windows

Drackonic, 2020-12-17:
Installed 6.3.3 from the releases and updated it without uninstalling the previous installation. The program will run until it reaches the loading screen before abruptly crashing and producing a windows .dmp file with the error:

Access violation - code c0000005

I have tried uninstalling and reinstalling from scratch, and disabling my antivirus to insure that it was not interfering. I can not seem to get it to fully load in. 

All drivers should be up to date as far as I'm aware.

Windows 10 64 bit,  Version 2004.

f4exb, 2020-12-17:
Sorry, it works. This is unrelated to SDRangel.

f4exb, 2020-12-17:
While it is not sure it has anything to do with version 6.3.3 vs earlier I have tested it against W10 version 1803.

Drackonic, 2020-12-18:
After much testing I eventually found and cleared the settings registry entry and this _seems_ to have solved the issue for me. Something I will note is that I updated from a much older version than I thought, going from 4.14.20 to the latest 6.3.3. Obviously this is conjecture but maybe the setting from an old version are incompatible and break things?

Drackonic, 2020-12-18:
And immediately after writing this I found something strange, selecting HackRF for sink devices seems to instantly crash the program.

Here is a video demonstrating the problem: https://www.youtube.com/watch?v=gvwdDNS_mzc

f4exb, 2020-12-18:
Yes sometimes depending on the context the current state that is saved at program exit from version X and automatically retrieved when launching program version Y may cause issues. This is more likely to be the case of a plugin that was used in version X and is no more compatible with the same in version Y. Since the retrieval of current state is automatic at startup you are trapped in a deadlock situation. I don't know how things are represented in Windows but in Linux it is enough to edit `~/.config/f4exb/SDRangel.conf` and remove the line that starts with `current=`.

The HackRF issue in Windows has already been reported in #718

Drackonic, 2020-12-19:
Ahh, that's interesting to know. It's a shame windows can't be so easy to do such things, at least not so far as I've seen. I was three reinstalls in when I decided to spend a while searching through the registry and eventually found the f4exb key and deleted the values within and that solved the issue for me. I imagine one of the conflicts could have been that I had several presets involving the HackRF.

I was not aware that was a known issue with it until now, thank you for informing me. I'll use the soapysdr workaround for it until it's able to be fixed.

Drackonic, 2020-12-20:
So been poking around trying to see what's up and even when the program starts its showing quite a lot of access violation errors in the windows debugger.

Here is an excerpt of it from WinDbg

(4234.3e84): Access violation - code c0000005 (first chance)
First chance exceptions are reported before any exception handling.
This exception may be expected and handled.
libusb_1_0!libusb_interrupt_transfer+0x65b4:
00007ffe`005a4e34 0fb611          movzx   edx,byte ptr [rcx] ds:00000000`00000000=??

(4234.3e84): Access violation - code c0000005 (!!! second chance !!!)
libusb_1_0!libusb_interrupt_transfer+0x65b4:
00007ffe`005a4e34 0fb611          movzx   edx,byte ptr [rcx] ds:00000000`00000000=??

It'll make a few of these before eventually starting.

Drackonic, 2020-12-20:
In WinDgb It produces this when you attempt to load the normal HackRF

``(1264.e90): Access violation - code c0000005 (first chance)
First chance exceptions are reported before any exception handling.
This exception may be expected and handled.
libusb_1_0!libusb_interrupt_transfer+0x18b0:
00007ffe`0b1f0130 488b8120020000  mov     rax,qword ptr [rcx+220h] ds:00000000`00000220=????????????????``

``(1264.e90): Access violation - code c0000005 (!!! second chance !!!)
libusb_1_0!libusb_interrupt_transfer+0x18b0:
00007ffe`0b1f0130 488b8120020000  mov     rax,qword ptr [rcx+220h] ds:00000000`00000220=????????????????``

Hopefully this is of any help at all.

closed 2021-01-24 by f4exb