f4exb/sdrangel #740: ADD NEW PLUG !

longyucheng123, 2020-12-23:
Hello! It is hoped that a plug-in can be added to the new version of the software to support the transmission of WAV files (windrm freedv DRM fdmdv sstv and other digital audio signals) that have been modulated.

f4exb, 2020-12-23:
Very useless tile and lack of research among existing plugins. The file source channel plugin achieves the same functionality although not taking .wav format. It is not very difficult to convert a .wav into a .sdriq if you just make the effort. However as .wav is popular the plugin could be made to accept stereo .wav files as the I/Q source.

longyucheng123, 2021-01-10:
Hello! USRP B210 hardware cannot be found after sdrangel software runs, but the hardware can be found through uhd_find_devices. How to set it? (Please see attached picture)发自我的华为手机-------- 原始邮件 --------发件人： Edouard Griffiths <notifications@github.com>日期： 2020年12月24日周四 00:21收件人： f4exb/sdrangel <sdrangel@noreply.github.com>抄送： longyucheng123 <longyucheng82@126.com>, Author <author@noreply.github.com>主    题： Re: [f4exb/sdrangel] ADD NEW PLUG ! (#740)
Very useless tile and lack of research among existing plugins. The file source channel plugin achieves the same functionality although not taking .wav format. It is not very difficult to convert a .wav into a .sdriq if you just make the effort. However as .wav is popular the plugin could be made to accept stereo .wav files as the I/Q source.

f4exb, 2021-01-11:
Not related to the subject

f4exb, 2021-03-16:
Too complex and not so useful because wav format can be transformed to sdriq easily. Dropped.

closed 2021-03-16 by f4exb