f4exb/sdrangel #752: DSD Demodulator requires audio sample rate of 48kHz

srcejon, 2021-01-18:
Hi,

I tried using the DSD Demod for the first time, but initially couldn't get it working (on Windows), despite fiddling with all of the demodulators settings. The problem turned out to be that it only works when the output audio sample rate is 48kHz and SDRangel on Windows seems to default to 8kHz for System default audio device. 

The documentation says the baseband sample rate needs to be at least 48kHz, but doesn't mention the output audio rate. Requiring a 48kHz baseband sample rate makes sense, but I wouldn't have expected their to be a requirement on the output audio rate. 

If there is, it perhaps should be in the docs and a warning could pop up if set too low? If not, perhaps there's a bug where audio sample rate is used in place of the channel sample rate, or something like that.

f4exb, 2021-03-16:
It depends on what you consider working or not. The base audio sample rate in the decoding subsystem is 8 kHz then there is an interpolator in integer rates that is able to upsample to an audio sample rate of 8, 16, 24, 32, 40, 48 kHz. If your audio sample rate does not fall on these values then the audio output may be distorted but this should not affect the decoding itself (meta data should still be decoded). 8 kHz should be fine with no interpolation applied. This is how it is designed and should normally work.

srcejon, 2021-03-16:
It wasn't a question of distorted audio, but the demod not being able to decode at all. 

I'll try to provide more detail on how to reproduce it, but have the radios setup for something else at the moment. New plugin on the way :)

f4exb, 2021-03-16:
OK don't spoil the surprise :) I think it should work by looking a the code but I haven't tried to reproduce the case either.

f4exb, 2021-04-19:
With audio rate 48 kS/s (OK):
[image]

With audio rate 8 kS/s (KO):
[image]

The diagram shows that when at 8k the waveform seen by the decoder is not correct. There is probably an unexpected side effect of the audio sample rate.

f4exb, 2021-04-19:
At 8k tweaking the baseband sample rate results in this:
```
2021-04-19 10:58:40.202 (D) DSDDemodSink::applyChannelSettings:  channelSampleRate:  12500  inputFrequencyOffset:  -200
```

At 48k
```
2021-04-19 10:59:37.217 (D) DSDDemodSink::applyChannelSettings:  channelSampleRate:  100000  inputFrequencyOffset:  -12700
```
Obviously 12500  Hz is wrong so it seems the audio sample rate setting spills over the setting of channel sample rate which should not be the case. It should always be 48k or more for the DSD case... So issue is not in  `DSDDemodSink` but somewhere in the upper layers.

f4exb, 2021-04-19:
First problem here when frequency shift changes (requested rate should be fixed 48000):
https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demoddsd/dsddemodbaseband.cpp#L154

Same here:
https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demoddsd/dsddemodbaseband.cpp#L138

And here...
https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demoddsd/dsddemodbaseband.cpp#L177

Thre reported issue corresponds to the latter but changing frequency or baseband sample rate when audio sample rate is not 48k will result in the same issue.

commit 5e7fbec1 (2021-04-19): DSD demod: request fixed 48000 S/s sample rate for the channel sample rate. Fixes #752
closed 2021-04-19 by f4exb