f4exb/sdrangel #754: SDRangel with SatPC32

Valstee, 2021-01-21:
Hi F4exb,

First a big thank you for your work !

I use SDRangel 6.5.0 to receive satellite telemetry with a LimeSDR mini. 
For tracking, I always used Gpredict, which works very well with SDRangel.
However, I would like to use SatPc32 to make my station autonomous while having a better control of my tracking parameters than with Gpredict. 

But I can't get SDRangel and SatPC32 to communicate...
SatPC32 uses a COM port to send a Doppler correction but the "RigCtl Server" feature listens to a TCP port.
To solve the problem, I tried a serial port emulator without success... 

Is there a solution for my problem ?

Thank you very much for your help,

Valentin from Sophia Antipolis ;-)

srcejon, 2021-01-22:
No, there's no option for serial port control currently. 

What do you mean by having better control of your tracking parameters?

Valstee, 2021-01-22:
Thanks for the answer !

For exemple, SatPC32 allows me to manage satellite tracking priorities when I activate the autotrack mode (via the modification of the Prior.SQF file -> q.v. http://www.dk1tb.de/manual_e.pdf)
Does Gpredict allow me to manage the autotracking mode?

Secondly, Gpredict's radio control regularly "disengages" without being able to fix the problem (after 20/30min). This is not the case with SatPC32.

srcejon, 2021-01-26:
Don't know about Gpredict.

With regards to the disconnection, it's possible the connection is being dropped as no data is sent for a while. Perhaps QAbstractSocket::KeepAliveOption needs to be set. I'll see if I can reproduce it.

Valstee, 2021-02-15:
I solved the problems with gpredict, just errors related to the administration rights of my computer.
Thanks a lot for your help @srcejon !

closed 2021-02-15 by Valstee