f4exb/sdrangel #759: Is sdrangel supporting the xtrx-SDR?

76nic, 2021-01-26:
Hi,
is sdrangel supporting the xtrx-SDR? If so, what library/interface is needed (e.g. gr-osmosdr)?
Many thanks!
Best regards,
76nic

f4exb, 2021-01-28:
It does:
  - https://github.com/f4exb/sdrangel/tree/master/plugins/samplesource/xtrxinput
  - https://github.com/f4exb/sdrangel/tree/master/plugins/samplesink/xtrxoutput
  - https://github.com/f4exb/sdrangel/tree/master/plugins/samplemimo/xtrxmimo

I am not sure it is available in the Ubuntu package because of the complexity of XTRX support libraries. It should work however if you compile from source. It is not available for Windows.

76nic, 2021-01-29:
Many thanks for your prompt reply! Yes, it seems that the deb package is not supporting the XTRX. By building from source you mean via docker? I played a bit with docker last night - it is not straight forward (at least for me ;-)). But I will dive into it a bit deeper. Hopefully, I will get this work in kubuntu kde.
The XTRX is a nice SDR but unfortunately the community is very small and library implementation in e.g. osmocom, python interface, ... is very slow.
Although I have not used sdrangel so much in the past, I follow it since years. One thing I have noticed is the active support and development! So - many thanks for all your efforts and great work !
Best regards,
76nic

f4exb, 2021-01-30:
Docker is not a must but will get you go faster if you are comfortable with Docker. You can also compile natively following these instructions: https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux

The downside is that instructions for manual build are not guaranteed to be up to date although they generally are while the Docker build is foolproof checked with the CI builds.

76nic, 2021-01-31:
Many thanks! I will try to build it according to that on a freshly installed Kubuntu 20.4 LTS to see if it works!
Again many thanks!
Best regards,
76nic

closed 2021-01-31 by 76nic