f4exb/sdrangel #760: Two-way remote audio + waterfall streaming with sdrangelsrv

mikaelnousiainen, 2021-01-27:
I'm interested in using sdrangelsrv as a backend for a simple web-based SDR application, which would expose the audio streams for RX and TX via WebRTC and use the new spectrum server feature for waterfall display.

This will obviously require a dedicated application with an API server and additional processing for audio + waterfall, but after reading some sdrangel plugin code I wasn't sure if extracting / ingesting external audio streams is possible in a headless sdrangelsrv. The spectrum server seems to cover the requirements for waterfall display.

The functionality I'm after seems to exist in some form in the audio sink plugin, as described here (UDP / RTP stream for audio):
https://github.com/f4exb/sdrangel/wiki/Networking-audio

However, I'm not sure if the audio sink plugin will require use of local PulseAudio devices (which wouldn't exist on a headless server instance). Additionally, it looks like there is no counterpart for UDP/RTP audio streaming in the source plugins.

Related questions:

1. Can the audio sink plugin be used for network audio streaming only (without local PulseAudio devices)? If not, could there be a separate "remote audio sink" plugin?
2. Would the same concept work as a "remote audio source" plugin, which would consume either raw UDP or RTP packets for transmitting audio on a TX-capable SDR, such as LimeSDR?
3. The waterfall spectrum server seems to have a hard-coded limit of outputting a maximum of 5 "frames" per second. Could this be made configurable, so that it is possible to control the speed of waterfall motion?

f4exb, 2021-01-28:
> 1. Can the audio sink plugin be used for network audio streaming only (without local PulseAudio devices)? If not, could there be a separate "remote audio sink" plugin?

There is no audio sink plugin so this does not refer to anything existing actually. However it is quite possible to route the audio output mix over the network following the mentioned instructions here: https://github.com/f4exb/sdrangel/wiki/Networking-audio This works also in multicast mode. I routinely use it to listen to the output of a  headless RPi in different places in the house on my mobile using MX Player. There is not much in place for the other way round to use audio from the network as a modulating source. However it could be possible to put something in place externally using a virtual audio input with Pulseaudio. 

Unlike what you say it is quite possible to use pulseaudio headless (I do it routinely on the RPi in headless mode) one just has to keep one terminal session open (same user as the user running the SDRangel instance). This is quite easy using a detached terminal with the "screen" command. Pulseaudio is a key component in the ecosystem and as mentioned it is very handy for creating virtual devices that can be used with SDRangel. 

Being able to route the individual audio output of channel sink plugins has some interest although I never found a use case for it. Usually I would scan and watch several channels at a time so I am more interested in the audio mix even if this gets messy sometimes.  The right place for this functionality is in each of the channel  sink plugins producing audio or in a common feature plugin (now that feature pliugins exist).

> 2. Would the same concept work as a "remote audio source" plugin, which would consume either raw UDP or RTP packets for transmitting audio on a TX-capable SDR, such as LimeSDR?

The use case of having a remote audio input as modulating source is worth enough to think of something more simple than piping audio externally. So this option could be investigated either directly in modulator plugins or through a common feature plugin. 

> 3. The waterfall spectrum server seems to have a hard-coded limit of outputting a maximum of 5 "frames" per second. Could this be made configurable, so that it is possible to control the speed of waterfall motion? 

the technology used for spectrum transmission is websockets. It will not work well at too high rates. 5 FPS may be a bit conservative but do not expect to have the same responsiveness as with the OpenGL GUI interface. It can be made configurable but there must be a hard limit and I don't see it higher than 10 FPS. If there is no limit I will receive issues about "remote spectrum does not work" because on average people who open this kind of issues are fairly ignorant about computer technology. Higher rates would require a dedicated UDP based technology (unlike TCP websockets) but websockets were chosen for convenience. It makes it also much easier for passing firewalls.

mikaelnousiainen, 2021-01-28:
@f4exb Thanks for your detailed answer!

Ok, so the audio output is normally a mix of the outputs of the demodulators? And each modulator can choose where to get its input -- and that pulseaudio is not the only input/output possible for the channel plugins? Am I understanding this correctly?

So the solution would be to attach network audio streams to individual modulators and demodulators via feature plugins -- can they attach to the inputs/outputs of channel plugins (and not only the sample source/sink directly)?

Optimally, there would be no need to use PulseAudio (or any other local audio functionality) on the server side, as the audio would be piped through network completely.

I do also understand the challenges with the spectrum server limitation, but removing the hard-coding would be the first step. The default value could be set to something around 5 anyway. Getting UDP packets from the spectrum server would definitely help, as I was researching use of WebRTC data channels for this use on the web browser side and it's all based on UDP anyway -- without the overhead of TCP.

f4exb, 2021-01-29:
Re spectrum server limitation: I do not remember right now how this was implemented but I think the rate would never make it past the GUI refresh rate that is capped at 20 FPS. I think it is triggered at a fraction of the GUI refresh rate. Giving access to the fraction number would be a way to control the spectrum rate at 20, 10, 6.67, 5, 4, ... FPS and is not too difficult to implement. Moreover it gives access to lower FPS which is in my opinion more useful for IoT like kind of applications.

Specific plugins for audio over network require a bit more of thinking.

Yes the output of modulators sharing the same audio device are mixed then optionally this mix can be sent over the network. The other way round does not exist but similarly for each modulator you can choose the audio input device.

f4exb, 2021-02-07:
Lifting the spectrum server limitation is even simpler than I thought since it is done in the `WSSpectrum` object and not in the caller of the `newSpectrum` method. Therefore it may be entirely controlled with the `WSSpectrum` parameters. Here is the piece of code with 200ms (5 FPS) hardcoded: https://github.com/f4exb/sdrangel/blob/master/sdrbase/websockets/wsspectrum.cpp#L166 

It can be possibly lifted off entirely and throttling can be controlled just with the spectrum averaging.

f4exb, 2021-02-07:
Concerning audio over network I will not make any change for the audio output since everything already exist in collaboration with pulseaudio. Pulseaudio is needed anyway for audio demods to work properly since it underlies the Qt audio interface. It is quite possible to make it work in a headless configuration as explained above. I have a Pi4 that has been working like this for weeks. With virtual audio devices it is also possible to isolate the output of one or several demods from the common audio mix.

mikaelnousiainen, 2021-02-07:
@f4exb Would you accept PRs if I work on this kind of functionality -> to have network-based and (PulseAudio-independent) output and input capabilities for audio? I'm not super familiar with the code base, but I think I could implement something like this if I know it would be accepted as a contribution.

f4exb, 2021-02-07:
> having PulseAudio as a part of this makes it extra complicated.

Can you please detail why.

f4exb, 2021-04-12:
Question answered.

closed 2021-04-12 by f4exb