f4exb/sdrangel #765: No sound until output changed

treaves, 2021-02-03:
I've just build from master.  My equipment:
- HackRF One with current firmware
- Ubuntu 20.04
- ThinkPad T470 (i5 four core)

I am tuned to a local NOAA station out of Cleveland.  For Channels, I've added a NFM Demodulator. When I start the SDR, the waterfall starts showing the signal. But I received no audio output. The console where I started from would show lots of
```
2021-02-03 15:08:06.434 (D) NFMDemodSink::feed: 0/1966 tail samples written
2021-02-03 15:08:06.477 (D) NFMDemodSink::feed: 0/1966 tail samples written
``
output.  I have to switch output device, then switch back (to the alsa_output) before I hear audio.  If I stop the SDR and restart it, I again have to switch / switch back the output.

f4exb, 2021-02-05:
Probably an issue with the audio setup in your system.

treaves, 2021-02-05:
The system is a standard Kubuntu setup. So it does have the inherited issues with sound a lot of Linux systems face; for example, it won't see the HDMI audio if the cable is attached after boot. Standard stuff.

On one test, sound did play at start, but it was very noisy; once I switched from the 'system' sound selection to the alsa-based one, it was better. 

The log output would seem to indicate that although the application is aware of my sound outputs, it's choosing the wrong one.

artwig, 2021-02-28:
@treaves, how are you installing?  I get similar behaviour when running in Docker, but not when using my own build of the deb.

F5UJA, 2021-07-09:
Hi, same problem here; Ubuntu 20.04.2.
All others SDR programs output audio without any issue (GQRX, CubicSDR)
I set audio Output and Intput on System default device
Any clue, on what else I should do?
Thanks,

closed 2021-12-20 by github-actions[bot]