f4exb/sdrangel #769: DeviceUISet::loadRxChannelSettings destroys GUI without calling setMessageQueueToGUI(nullptr)

srcejon, 2021-02-07:
I've noticed SDRangel crashing (on Windows) when trying to load a preset, when a PacketDemodulator exists.

This appears to be because the PacketDemod::handleMessage can send messages to the GUI, and DeviceUISet::loadRxChannelSettings isn't calling setMessageQueueToGUI(nullptr) before it destroys the GUI. 

Even though it destroys the demod on the following line, it seems  PacketDemod::handleMessage can be called with a DSPNotification in between, resulting in handleMessage trying to send the message to a GUI that's been destroyed.

f4exb, 2021-02-07:
`handleMessage` should not try to send a message to the GUI if the GUI pointer is null so it may be a case of the GUI pointer not properly initialized at construction time. I'll have a look...

srcejon, 2021-02-07:
Yes, there is a check for nullptr. The problem, I think, is that the pointer isn't being set to nullptr. This is not at construction time, but just before the GUI is destroyed.

Here's an example, in mainwindow.cpp, when GUIs are destroyed, you have:

        m_deviceUIs.back()->m_deviceAPI->getSampleSource()->setMessageQueueToGUI(nullptr); // have source stop sending messages to the GUI
        m_deviceUIs.back()->m_deviceGUI->destroy();

Which prevents messages being sent to a destroyed GUI.

However, this isn't the case in DeviceUISet::loadRxChannelSettings. It just calls destroy without having called setMessageQueueToGUI(nullptr) first. See:

        // clear list
        for(int i = 0; i < m_channelInstanceRegistrations.count(); i++)
        {
            qDebug("DeviceUISet::loadRxChannelSettings: destroying old channel [%s]",
                qPrintable(m_channelInstanceRegistrations[i].m_channelAPI->getURI()));
            m_channelInstanceRegistrations[i].m_gui->destroy(); // FIXME: stop channel before
            m_channelInstanceRegistrations[i].m_channelAPI->destroy();
        }

I think we need:

        for(int i = 0; i < m_channelInstanceRegistrations.count(); i++)
        {
            qDebug("DeviceUISet::loadRxChannelSettings: destroying old channel [%s]",
                qPrintable(m_channelInstanceRegistrations[i].m_channelAPI->getURI()));
            m_channelInstanceRegistrations[i].m_channelAPI->something->setMessageQueueToGUI(nullptr)();
            m_channelInstanceRegistrations[i].m_gui->destroy(); // FIXME: stop channel before
            m_channelInstanceRegistrations[i].m_channelAPI->destroy();
        }

Or perhaps stop the channel as per the FIXME comment, if that stops message handling.

commit f541b534 (2021-02-08): Set channel message queue to null in channel before destroying its GUI. Fixes #769
closed 2021-02-08 by f4exb