f4exb/sdrangel #771: Spectrum GUI: autoscaling

f4exb, 2021-02-09:
Following #768 this ticket is to implement spectrum top level and range autoscaling. This will be done one time with a push button.

srcejon, 2021-02-10:
Lovely thanks.

One more suggesting having tried it - it might be worth setting the reference level to some % of the range or a fixed amount of db above the calculated peak, so there are generally a few pixels of blank space. This would mean you can easily be sure what the level of the peak is. When the peak is right at the top of the display, it could be that it has increased beyond the reference level, so you have to adjust the ref level to see:

E.g. currently autoscale results in:

[image]

But this might be a little clearer:

[image]

f4exb, 2021-02-10:
Well this is quite heuristic in fact and it will be difficult to find one size that fits all. Anyway the idea is to take the mean of the n largest and n lowest values as a base then the top is 10 times the maximum average which moves it +10dB towards the top already (but keep in mind this is an average) then for the bottom of the range it is 2 times the average of the minima which pulls it +3dB towards the top also. Now a few things could be changed. The factor n is the fft size divided by 32 which for 1k FFT yields 32 maybe this is too much and if you have 2 prominent signals as in your picture it will fall roughly in the middle of the peaks. The final "moving" factor can be changed also.

However I think the point is to make the signal fall roughly in range in most cases. Some further adjustments are inevitable.

srcejon, 2021-02-10:
The averaging makes sense to me for the lower limit (because you don't want to see all of the noise - and this looks spot on), but I would have thought you'd use the highest peak (+ a bit) for the upper limit, as you generally want to see all of the signals. Maybe I'm just too used to spectrum analysers that don't work properly when the signals clip!

Much appreciated, anyway.

f4exb, 2021-02-10:
Yes retaining just the maximum appears to make things nicer. Also you may engage the fixed average to stabilize signals before pushing the auto range and then come back to no average (or other average styles). If you really want to catch the maximum on pulsed signals you may as well use the max instead of the fixed average.

srcejon, 2021-02-11:
Yep. Like that, thanks!

f4exb, 2021-02-13:
Done in v6.5.4

closed 2021-02-13 by f4exb