f4exb/sdrangel #773: Spectrum GUI: frequency zooming

f4exb, 2021-02-15:
This ticket is opened following #768 to handle the specific case of frequency zooming. That is zooming the FFT along the frequency (x) axis. This will be a GUI only feature using mouse wheel and modifier (mouse wheel is used to move channels already).

This does not cover the proper handling of "manual" markers. This will be covered in another ticket. For now it is not expected that these markers will follow the zooming in any way. In fact these markers should be promoted to proper frequency markers like the channel markers.

f4exb, 2021-02-17:
Now on master. Use Alt+Mousewheel up and down for zoom in / zoom out. Use Alt+click for moving frequency to the center (panning). If you use LxQt you may have to disable the Alt+Mouse bindings in `~/.config/openbox/lxqt-rc.xml` then do `openbox --reconfigure`

f4exb, 2021-02-17:
Build issues:
In MSVC 2019:
```
[00:16:12] glspectrum.cpp.obj : error LNK2001: unresolved external symbol "public: virtual bool __cdecl SpectrumVis::MsgFrequencyZooming::matchIdentifier(char const *)const " (?matchIdentifier@MsgFrequencyZooming@SpectrumVis@@UEBA_NPEBD@Z)
```
in Ubuntu 20.04 due to Qt version 5.12.8 not supporting yet the `position` method of `QWheelEvent`:
```
[00:17:13] ../sdrgui/gui/glspectrum.cpp:2281:28: error: ‘class QWheelEvent’ has no member named ‘position’; did you mean ‘posted’?
[00:17:13]   const QPointF& p = event->position();
[00:17:13]                             ^~~~~~~~
[00:17:13]                             posted
```

srcejon, 2021-02-17:
Thanks Edouard! 

Is ALT necessary, as the scrollwheel doesn't appear to be used for anything else? Or do you have something else planned? (Could also do y-axis zoom as well, I guess (i.e. simultaneously adjust reference level / range))

f4exb, 2021-02-18:
> the scrollwheel doesn't appear to be used for anything else? 

yes it is used to move channels (maybe this is undocumented) however this happens only when mousing over the central line so the conflict is only in this condition. If you manage by priority then zooming could be active only when the mouse is not over the central line of a channel. Maybe this is acceptable. For frequency centering a keyboard modifier is necessary at least when there is only one channel since in this case the left click is used to move the unique channel at mouse pointer position.

> Could also do y-axis zoom as well

It makes things a bit tricky and then you must use a modifier to tell the difference between X and Y axes. It is not strictly necessary as as you mention you already have reference level and range controls. However it would make sense that autoscaling is done only over the zoomed (visible) part I am not sure this is the case.

f4exb, 2021-02-18:
Simultaneously adjusting reference level / range via pointer + mousewheel may be possible while the pointer is inside the power scale. That would make sense and would not interfere with the spectrum area. I have to experiment a bit with that and possibly open a new ticket as this is unrelated to frequency scaling.

srcejon, 2021-02-18:
Good idea - I think that's a little nicer than using the two spin boxes, as it's just a single action.

srcejon, 2021-02-18:
BTW, the ChannelAnalyzer crashes when I open it (on Windows) - I'm guessing it's related to these changes.

I get an exception in glspectrum.cpp line 2644 in formatTextInfo()

f4exb, 2021-02-18:
It is a zero divide. Things are probably not initialized the same way in the Channel Analyzer:
```
2644            info.append(tr("CF:%1 ").arg(displayScaled(centerFrequency, 'f', getPrecision(centerFrequency/frequencySpan), true)));
[Current thread is 1 (Thread 0x7fa4140ba000 (LWP 2773638))]
(gdb) p frequencySpan
$1 = 0
```

f4exb, 2021-02-18:
There is a bad initialization of the time scale as well that persists and that I have noticed only occasionally on some plugins.

f4exb, 2021-02-20:
Implemented in v6.5.5

closed 2021-02-20 by f4exb