f4exb/sdrangel #774: Spectrum markers resolution

srcejon, 2021-02-15:
Hi,

It seems the spectrum markers have a resolution lower than the spectrum. For example, at low sample rates, the delta can read 0, even when surrounding a peak. E.g:

[image]

If you increase the delta - it seems the next step up from 0 is 16Hz, then 32Hz.

(This example is obviously just a test case - actual usage was trying to measure a small amount of Doppler shift from the Moon)

f4exb, 2021-02-16:
X axis to frequency correspondence is done via the frequency scale: https://github.com/f4exb/sdrangel/blob/master/sdrgui/gui/glspectrum.cpp#L2057

Screen resolution has probably a role to play here.

f4exb, 2021-02-16:
When the zooming is in place (soon...) you will be able to get something like this:
[image]

srcejon, 2021-02-16:
Nice :)

f4exb, 2021-02-17:
Now on `master` see #773

f4exb, 2021-04-10:
Enhancing spectrum markers as real markers like channel markers attached to a frequency and moving with it will be covered as one of the topics of #768. Hence this one is closed in favor of #768

closed 2021-04-10 by f4exb