f4exb/sdrangel #775: sdrangel and llvm 11 error

fedya, 2021-02-16:
Trying to build sdrangel with llvm and got this

```
In file included from /home/omv/sdrangel/BUILD/sdrangel-6.5.4/build/pluginssrv/channeltx/remotesource/remotesourcesrv_autogen/EWIEGA46WW/../../../../../../plugins/channeltx/remotesource/remotesource.h:21:
In file included from /usr/include/qt5/QtCore/QObject:1:
In file included from /usr/include/qt5/QtCore/qobject.h:46:
In file included from /usr/include/qt5/QtCore/qobjectdefs.h:48:
In file included from /usr/include/qt5/QtCore/qnamespace.h:43:
/usr/include/qt5/QtCore/qglobal.h:1052:9: error: incompatible integer to pointer conversion assigning to 'QDebug::Stream *' from 'long'
    t = std::forward<U>(newValue);
        ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/qt5/QtCore/qdebug.h:118:46: note: in instantiation of function template specialization 'qExchange<QDebug::Stream *, long>' requested here
    QDebug(QDebug &&other) noexcept : stream{qExchange(other.stream, nullptr)} {}
        |       |       |       |       |    ^
1 error generated.
```

fedya, 2021-02-16:
I think issue related not to llvm, probably culprit is sdrangel itself

f4exb, 2021-02-16:
Issue is in Qt and its compatibility with LLVM on that point. This is unresolvable. It works with the compilers used for the building the artifacts: gcc in Linux and MSVC 2019 in Windows.

berolinux, 2021-02-17:
The problem is not with LLVM, the problem is with cm256cc (and it is reproducable with current versions of gcc as well).

This is what's going on:

sdrangel's plugins/channelltx/remotesource/remotesourcesource.h #includes cm256cc/cm256.h before including QObject.
cm256.h #includes gf256.h, which includes an (incorrect) workaround for being used with ancient compilers:
```
#ifndef nullptr
    #define nullptr NULL
#endif
```
Since nullptr is not #defined anywhere (it's a part of the language, not a #define), the workaround triggers even with compilers that know about nullptr.
So nullptr is defined to NULL, which in turn is defined to 0L.

QObject then does `qExchange(other.stream, nullptr)` after that bogus redefinition of nullptr, making it `qExchange(other.stream, 0L)`, correctly making standards compliant compilers scream about trying to swap a pointer with an integer.

The correct fix is simply removing the bogus definition of nullptr from cm256cc.

berolinux, 2021-02-17:
This is already fixed in cm256cc git master
https://github.com/f4exb/cm256cc/commit/c0e92b92aca3d1d36c990b642b937c64d363c559
but it hasn't made it into any cm256cc release yet.

f4exb, 2021-02-17:
Normally one would rather rely on the top of the master branch rather than tags but I can make an update in cm256cc. Anyway the compilation instructions explicitly mentions this commit for cm256cc:
https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux#cm265cc

So one who follows instructions should not have the issue. It also explains why all the builds work ...

fedya, 2021-02-17:
@f4exb i packaged it for OpenMandriva linux distro and only in a rare cases i can pack master branch of package instead of release.
Anyway, now it's fixed. Thanks.

closed 2021-02-17 by fedya