f4exb/sdrangel #777: DSD Digital demodulator not in the released version

programmin1, 2021-02-17:
I upgraded 20.04, installed your .deb release, and I see no DSD Digital demodulator? Why is this not included in the binary release?

f4exb, 2021-02-18:
It is ...

programmin1, 2021-02-18:
In the .deb for 20.04? I must be missing something...

f4exb, 2021-02-18:
You can see for yourself and unpack the .deb The build listing also shows that libdemoddsd.so is rightfully produced: https://ci.appveyor.com/project/f4exb/sdrangel/builds/37757365/job/9ryi273h95pcydy4

srcejon, 2021-02-18:
Maybe check the log file when the plugins are loaded. If a shared library is missing, the plugin may not be visible.

f4exb, 2021-02-18:
well `libdsdcc` and `libmbe` are there too I think they are the only specific to DSD demod.

f4exb, 2021-02-18:
`libmbe` is there yet there is something fishy. Tried inside a 20.10 VM:
```
2021-02-18 17:39:04.954 (W) PluginManager::loadPluginsDir: Cannot load library /usr/lib/sdrangel/plugins/libdemoddsd.so: (libmbe.so.1: cannot open shared object file: No such file or directory)
```

f4exb, 2021-02-18:
Maybe it has stg to do with the cascading dependency `libdsdcc` -> `libmbe`.
```
ldd /usr/lib/sdrangel/libdsdcc.so
        linux-vdso.so.1 (0x00007fff06dee000)
        libmbe.so.1 => not found
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fc431b8d000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fc431a3e000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fc431a23000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fc431839000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fc431dcb000)
```
And there is no runtime path set for `libdsdcc`

f4exb, 2021-02-18:
Maybe adding `mbelib` here would help: https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demoddsd/CMakeLists.txt#L61 
Else I have no idea...

f4exb, 2021-02-20:
This one should work: https://ci.appveyor.com/api/buildjobs/oru3ujsb138rbij5/artifacts/sdrangel-690-master.tar.gz

programmin1, 2021-02-21:
Hmm I just re installed that and I must be missing something...

[image]

f4exb, 2021-02-22:
It seems it works or not depending on installation details however I see something wrong here (in the case it does not work):
```
chrpath /usr/lib/sdrangel/libdsdcc.so
/usr/lib/sdrangel/libdsdcc.so: RUNPATH=/home/appveyor/projects/sdrangel/obj-x86_64-linux-gnu/external/lib:::::::::::::::
```
This path is from the build environment not the final installation path which conversely is set correctly for the plugin:
```
chrpath /usr/lib/sdrangel/plugins/libdemoddsd.so 
/usr/lib/sdrangel/plugins/libdemoddsd.so: RUNPATH=/usr/lib/sdrangel
```

f4exb, 2021-02-22:
Indeed:
```
sudo chrpath -r /usr/lib/sdrangel  /usr/lib/sdrangel/libdsdcc.so
```
Fixes the issue but I would like it to work right away in the CI (Appveyor).

f4exb, 2021-04-13:
Sorry this is not possible to fix.

madarexxx, 2023-08-12:
Hi there :) 
I have found a reason why this .so load fails. The libdsdcc.so.1 file does not have correct library search path, since it is screwed by CI/CD process. You can check it  as follows:

```
readelf -d /usr/lib/sdrangel/libdsdcc.so.1

Dynamic section at offset 0x37af0 contains 30 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libmbe.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x000000000000000e (SONAME)             Library soname: [libdsdcc.so.1]
 0x000000000000001d (RUNPATH)            Library runpath: [/home/appveyor/projects/obj-x86_64-linux-gnu/external/lib:::::::::::::::]
 0x000000000000000c (INIT)               0xe000
```

The solution is to patch .so file as follows (obviously I have patched my installed from .deb version):
`sudo patchelf --set-rpath /usr/lib/sdrangel/ /usr/lib/sdrangel/libdsdcc.so.1`

Would you mind adding this to release binary build process please? This is very frustrating bug for a new user, since it is not mentioned anywhere in docs for installation and DSD decoder. 

Thanks for your software, I love RSP1 native support and very advanced demodulator selection

crok, 2024-02-22:
This is still an issue (using Manjaro && AUR package). The [workaround here](https://github.com/f4exb/sdrangel/issues/777#issuecomment-781476539) did the trick though.

srcejon, 2026-08-17:
Reopened as still an issue.

It appears to be due to the fact that in external/CMakeLists.txt, we are just copying built shared libraries rather than installing them. (INSTALL_COMMAND is "" for many of the external projects) and so RPATH isn't handled properly. Sorting that out is a fairly large change.

closed 2026-08-17 by f4exb