f4exb/sdrangel #778: Original HackRF Support broken (Segmentation Fault)

TGCFabian, 2021-02-17:
Since the introduction of SoapySDR within SDRAngel, the original HackRF Support has been broken
Ive been using SDRAngel on version 4.21.3 untill upgrading today
It took me 2 days untill i figured out why its happening

When trying to use official HackRF Support, within console all it prints out is "Segmentation Fault"
Personally im not good at debugging, so thats all i'll be able to provide as of now

This small bug might cause confusion for HackRF Only users that are upgrading to the new version

This has also been mentioned in: https://github.com/f4exb/sdrangel/issues/718

f4exb, 2021-02-18:
> It took me 2 days untill i figured out why its happening

If the only piece of information you got is 'segmentation fault" then you did not really figured out why this is happening. Do you have more information? How does this relate to SoapySDR?

Also you mention #718 but this is specific to Windows. Are you running on Windows?

I personally have no issues with HackRF on Ubuntu 20.10

TGCFabian, 2021-02-18:
Its windows specific, happens on several windows machine of mine, all windows 10 based

Sorry for not mentioning that
Any other software worked fine with the HackRF, it works as long as i use SoapySDR

f4exb, 2021-02-18:
OK so this is a dupe of #718. If you have any more comments in particular if you did find something new please add your comments to #718.

closed 2021-02-18 by f4exb