f4exb/sdrangel #779: Spectrum GUI: power and time zooming (Y zoom)

f4exb, 2021-02-18:
This ticket is opened following #768 to handle the specific case of power (histogram) or time (waterfall) zooming. That is zooming  along the y axis.

Power zooming acts on reference level and range while time zooming acts on FFT overlap. All these controls exist already in the UI this is just a convenience.

f4exb, 2021-02-20:
Implemented in v6.5.5

closed 2021-02-20 by f4exb

srcejon, 2021-02-22:
Thanks!

One little bit of feedback, it seems that if acquisition is stopped, the Y-axis zoom updates the ref level / range spin boxes immediately, but (on Windows at least) the spectrum itself doesn't update until the cursor is moved out of the spectrum area.  When acquisition is running, then it appears to be updated immediately (but presumably is due to new data being plotted).