f4exb/sdrangel #780: SoapySDR file sink not recording.

Drackonic, 2021-02-19:
Using HackRF with the SoapySDR workaround, when selecting file sink for reception it is unable to record in any destination or any radio settings. No file is created and the program does not appear to attempt to start the recording ( no time, file size, ect.)

SDRangel 6.5.4
Windows 10 2004 64bit.

f4exb, 2021-02-19:
> it is unable to record in any destination

Can you be more specific? How do you specify the file name?

Drackonic, 2021-02-19:
I used the select output file button in the interface of the file sink channel plugin to select a destination and name. In this case I set the name as test, and the destination as the SDRangel root directory.

f4exb, 2021-02-19:
The readme here: https://github.com/f4exb/sdrangel/tree/master/plugins/channelrx/filesink#introduction describes how the final file name is composed. It does not mention however that the extension `.sdriq` must be given. This is not too convenient however and this extension could be added implicitely if missing and then the described rules applied.

See how this may also fit with the SigMF file sink.

commit b865d995 (2021-02-19): FileSink: give .sdriq extension to provided file name automatically. Fixes #780
closed 2021-02-19 by f4exb