f4exb/sdrangel #783: Presets appear to be saved in alphabetical order, rather than order in which they are created

srcejon, 2021-02-22:
It seems presets have the channels saved in alphabetical order, rather than the order in which they are created. For example:

- Add NFM Demod
- Add AM Demod
- Save preset
- Load preset
- Channels will be swapped and appear as AM Demod then NFM Demod

While generally not a problem - where it can be a little confusing is if you are controlling the channels via the API, as channel indexes will change. 

Should this behaviour be changed? Or is it just something that needs to considered when using the API?

f4exb, 2021-02-22:
Maybe I need to double check but if I remember right in the presets channels are sorted by frequency shift then by alphabetical order if their identifers (i.e. AMDemod comes before NFMDemod). When you create new channels in the session they are just appended. When importing channels from a preset there is the possibility to have prior knowledge of the order or make a GET query to list the channels and their respective indexes.

srcejon, 2021-02-22:
Ok - I'll update my code to deal with it. Thanks.

closed 2021-02-22 by srcejon