f4exb/sdrangel #784: DATV: display MER and possibily CNR

f4exb, 2021-02-22:
This issue is opened following comments on #18 to implement display of MER and possibly CNR in the GUI. The data appears to be available but its display has never been implemented.

darksidelemm, 2021-02-27:
I've seen the new meters for this, and they look great! 
Is there any chance of getting a windows build for this soon? We're doing a balloon launch next weekend, and we have a lot of stations setup using SDRangel on Windows ready to receive. It would be great if they were able to report back on MER levels during the flight!

f4exb, 2021-02-27:
The link to the Windows build in Appveyor is here: https://ci.appveyor.com/project/f4exb/sdrangel/builds/37902890/job/148ifipx45bpym7l/artifacts

darksidelemm, 2021-02-27:
Brilliant, thanks!

f4exb, 2021-02-28:
Implemented in v6.6.0

closed 2021-02-28 by f4exb