f4exb/sdrangel #786: Detection of SoapySDR does not work

artwig, 2021-02-27:
It doesn't look like SoapySDR is detected properly, the following seems to work for me but I have no idea if it is correct.

```diff
diff --git a/cmake/Modules/FindSoapySDR.cmake b/cmake/Modules/FindSoapySDR.cmake
index d52c3432e..13f745b9b 100644
--- a/cmake/Modules/FindSoapySDR.cmake
+++ b/cmake/Modules/FindSoapySDR.cmake
@@ -9,7 +9,7 @@ if(NOT SOAPYSDR_FOUND)
     set(${VERSION} "${CMAKE_MATCH_1}" PARENT_SCOPE)
   endfunction(_SOAPY_SDR_GET_ABI_VERSION)
 
-  pkg_search_module (LIBSOAPYSDR_PKG soapysdr>=0.4.0 SoapySDR>=0.4.0)
+  pkg_check_modules (LIBSOAPYSDR_PKG soapysdr>=0.4.0 SoapySDR>=0.4.0)
 
   if(NOT LIBSOAPYSDR_PKG_FOUND OR (DEFINED SOAPYSDR_DIR))
```

Soapy is installed using the package in Ubuntu 20.04:
```
ii  libsoapysdr-dev:amd64 0.7.2-1      amd64        SoapySDR library development files
```

Before the patch, it doesn't say if it worked or not:
```
-- Checking for one of the modules 'soapysdr>=0.4.0;SoapySDR>=0.4.0'
```

After the patch:
```
-- Checking for modules 'soapysdr>=0.4.0;SoapySDR>=0.4.0'
--   No package 'soapysdr' found
-- Found SoapySDR: /usr/include, /usr/lib/x86_64-linux-gnu/libSoapySDR.so
```

f4exb, 2021-02-27:
The change made to replace `pkg_check_modules` by `pkg_search_module` was made to fix #519. Thus this would be a regression.

artwig, 2021-02-28:
Actually, the if statement after the `pkg_check_modules` seems suspect.  It works fine if I remove it like this, which seems consistant with the others:
```diff
diff --git a/cmake/Modules/FindSoapySDR.cmake b/cmake/Modules/FindSoapySDR.cmake
index d52c3432e..ea8c0b0f1 100644
--- a/cmake/Modules/FindSoapySDR.cmake
+++ b/cmake/Modules/FindSoapySDR.cmake
@@ -11,42 +11,38 @@ if(NOT SOAPYSDR_FOUND)
 
   pkg_search_module (LIBSOAPYSDR_PKG soapysdr>=0.4.0 SoapySDR>=0.4.0)
 
-  if(NOT LIBSOAPYSDR_PKG_FOUND OR (DEFINED SOAPYSDR_DIR))
-
-    find_path (SOAPYSDR_INCLUDE_DIR
-      NAMES SoapySDR/Version.h
-      HINTS ${SOAPYSDR_DIR}/include
-            ${LIBSOAPYSDR_PKG_INCLUDE_DIRS}
-      PATHS /usr/include
-            /usr/local/include
-      )
-
-    find_library (SOAPYSDR_LIBRARY
-      NAMES SoapySDR
-      HINTS ${SOAPYSDR_DIR}/lib
-            ${SOAPYSDR_DIR}/lib64
-            ${CMAKE_INSTALL_PREFIX}/lib
-            ${CMAKE_INSTALL_PREFIX}/lib64
-            ${LIBSOAPYSDR_PKG_LIBRARY_DIRS}
-      PATHS /usr/local/lib
-            /usr/local/lib64
-            /usr/lib
-            /usr/lib64
-      )
-
-    if (SOAPYSDR_INCLUDE_DIR AND SOAPYSDR_LIBRARY)
-      set(SOAPYSDR_FOUND TRUE)
-      # get the root of SoapySDR; used on cpack
-      string(REGEX REPLACE "/lib/.*${CMAKE_SHARED_LIBRARY_SUFFIX}" "" SOAPYSDR_ROOT ${SOAPYSDR_LIBRARY})
-      # get the soapy version; to using FindPkgConfig because we can use SOAPYSDR_DIR
-      _SOAPY_SDR_GET_ABI_VERSION(SOAPYSDR_ABI_VERSION ${SOAPYSDR_INCLUDE_DIR})
-      message (STATUS "Found SoapySDR: ${SOAPYSDR_INCLUDE_DIR}, ${SOAPYSDR_LIBRARY}")
-    else()
-      message (STATUS "SoapySDR not found")
-    endif()
-
-    mark_as_advanced(SOAPYSDR_INCLUDE_DIR SOAPYSDR_LIBRARY SOAPYSDR_ROOT SOAPYSDR_ABI_VERSION)
-
-  endif(NOT LIBSOAPYSDR_PKG_FOUND OR (DEFINED SOAPYSDR_DIR))
+  find_path (SOAPYSDR_INCLUDE_DIR
+    NAMES SoapySDR/Version.h
+    HINTS ${SOAPYSDR_DIR}/include
+          ${LIBSOAPYSDR_PKG_INCLUDE_DIRS}
+    PATHS /usr/include
+          /usr/local/include
+    )
+
+  find_library (SOAPYSDR_LIBRARY
+    NAMES SoapySDR
+    HINTS ${SOAPYSDR_DIR}/lib
+          ${SOAPYSDR_DIR}/lib64
+          ${CMAKE_INSTALL_PREFIX}/lib
+          ${CMAKE_INSTALL_PREFIX}/lib64
+          ${LIBSOAPYSDR_PKG_LIBRARY_DIRS}
+    PATHS /usr/local/lib
+          /usr/local/lib64
+          /usr/lib
+          /usr/lib64
+    )
+
+  if (SOAPYSDR_INCLUDE_DIR AND SOAPYSDR_LIBRARY)
+    set(SOAPYSDR_FOUND TRUE)
+    # get the root of SoapySDR; used on cpack
+    string(REGEX REPLACE "/lib/.*${CMAKE_SHARED_LIBRARY_SUFFIX}" "" SOAPYSDR_ROOT ${SOAPYSDR_LIBRARY})
+    # get the soapy version; to using FindPkgConfig because we can use SOAPYSDR_DIR
+    _SOAPY_SDR_GET_ABI_VERSION(SOAPYSDR_ABI_VERSION ${SOAPYSDR_INCLUDE_DIR})
+    message (STATUS "Found SoapySDR: ${SOAPYSDR_INCLUDE_DIR}, ${SOAPYSDR_LIBRARY}")
+  else()
+    message (STATUS "SoapySDR not found")
+  endif()
+
+  mark_as_advanced(SOAPYSDR_INCLUDE_DIR SOAPYSDR_LIBRARY SOAPYSDR_ROOT SOAPYSDR_ABI_VERSION)
 
 endif(NOT SOAPYSDR_FOUND)
```

It seems if `pkg_search_modules` works, it needs to do the rest to fully detect it, but doesn't because `LIBSOAPYSDR_PKG_FOUND` is true.  Incidentally, commenting out the `pkg_search_modules` also works, as does passing `-DSOAPYSDR_DIR=/usr`.

f4exb, 2021-03-04:
I do not intend to change the present cmake files and therefore this issue can be closed.

closed 2021-03-04 by f4exb