f4exb/sdrangel #789: Distorted audio when sample rate is 44.1kHz

srcejon, 2021-03-01:
This zip file contains a sdriq file of a 1k sine wave sampled at 48000 Hz

[sin.zip](https://github.com/f4exb/sdrangel/files/6063233/sin.zip)

To recreate distortion:

- Set audio output rate to 44100 via Preferences > Audio
- Set FileInput as source and select this file
- Add SSB Demodulator
- Hit play

Distortion is audible and can be seen in the spectrum. Same on both Linux and Windows.

Strangely, sample rate of 44000 sounds fine. (22050 is distorted whereas 22000 is also fine).

f4exb, 2021-03-16:
Not able to reproduce:
[image]

[image]

srcejon, 2021-03-16:
Hi. It's definitely reproducible for me. A couple of things to try:

Look at the spectrum in SSB demod rather than waterfall. 

Here is clean spectrum at 48000:

[image]

But here is the distortion at 44100:

[image]

At 44000, there is a harmonic, but that sounds ok:

[image]

Here is how it sounds (these contain .wav files):

[44100.zip](https://github.com/f4exb/sdrangel/files/6147467/44100.zip)

[44000.zip](https://github.com/f4exb/sdrangel/files/6147474/44000.zip)

One thing I've noticed, is that it does not occur if the sample rate is set after the SSB demod is already open. It needs to be set to 44100 before opening the SSB demod.

f4exb, 2021-03-16:
Well:
 - there is a glitch at every ~9s because of the looping this is normal
 - the distortion on the spectrum at 44100 is about -70 dBc which I think is acceptable. This is probably due to precision errors in the interpolation
 - at 44000 the secondary peak is even a bit larger but still below ~60 dBc

It is difficult to tell the difference between the two .wav files because they are both overdriven. I just figured out that if you do not engage AGC then the 0 dB setting on the volume is still too much. You may try the latest commit on master https://github.com/f4exb/sdrangel/commit/996e4d7816621ed99a23eb3bb2df49c72c6dab9a that extends the volume range from -10 to +40 dB. I found that at about -6 dB it starts to be OK.

srcejon, 2021-03-16:
Yep, thanks. With the lower gain the distortion is no longer audible. (I thought -60dBc should be fine, but could clearly hear it - didn't realise the problem was the overdrive)

Just noticed another possible smaller issue:

- Set audio sample rate to 48k
- Open SSB demod
- Start playing sine wave
- While playing, set audio sample rate to 44100
- After a couple of loops of the file, there's a morse sounding effect and in the log lots of messages of the form:

SSBDemodSink::feed: 0/512 tail samples written

f4exb, 2021-03-16:
The audio sample rate is possibly changed only when the device audio name changes. Audio definitions are supposed to be relatively static and it would be rather difficult to implement something more dynamic.

f4exb, 2021-04-23:
Warning added here: https://github.com/f4exb/sdrangel/wiki/Audio-management

I think now this can be closed.

closed 2021-04-23 by f4exb