f4exb/sdrangel #791: Doppler / input frequency offset adjustment

srcejon, 2021-03-02:
The Doppler correction in the Satellite Tracker adjusts selected channel's input frequency offset. However, this can currently result in a brief, but noticeable, disruption to to the audio, so it isn't possible make frequent small corrections.

This can be heard without the Satellite Tracker, just by continuously adjusting the input offset in the SSB demod.  (I did briefly look at adjusting the centre frequency instead, but that has a similar sounding result). I haven't looked in to the code, but I'd guess some samples might be lost when the filters are reconfigured. Perhaps there's a way this adjustment could be more seemless? It would make regular SSB frequency scanning sound a little nicer too.

f4exb, 2021-03-03:
I have played a little bit with the code. There is a mutex lock when any setting is changed but this does not seem to be the issue. The real issue is here: https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demodssb/ssbdemodbaseband.cpp#L151

As the comment says `applyAudioSampleRate` is invoked "in case" the channel sample rate changes but this occurs relatively rarely only if moving the frequency cross some boundaries where the half band decimation chain needs to be changed.

the `applyAudioSampleRate` causes the redefinition of the audio interpolation filter and this is what causes discontinuity in audio:
```
2021-03-03 00:42:54.482 (D) SSBDemodSink::applyAudioSampleRate: 48000
2021-03-03 00:42:54.482 (D) Interpolator::createPolyphaseLowPass: ntaps: 32
```
So the process should be refined in some way so that effectively changing the interpolator occurs only if really necessary.

f4exb, 2021-03-03:
Applies to:
  - amdemodbaseband.cpp
  - bfmdemodbaseband.cpp
  - dsddemodbaseband.cpp
  - nfmdemodbaseband.cpp
  - ssbdemodbaseband.cpp
  - vordemodscbaseband.cpp
  - wfmdemodbaseband.cpp

srcejon, 2021-03-03:
Thanks - that sounds much better.

f4exb, 2021-03-03:
Implemented in v6.6.1

closed 2021-03-03 by f4exb