f4exb/sdrangel #793: With No Limit FPS Waterfall/Spectrum freezes until cursor move out

Cherenkov11, 2021-03-03:
I noticed that when I set the FPS limit to NL and you zoom in/out on the spectrum it freezes until you move the cursor out of the waterfall and something similar happens when you change FFT size, but this time you need to move the cursor over the waterfall to defrost it.

SDRangel: 6.6.0
OS: Archlinux

I can reproduce it with nouveau driver and nvidia driver on different PCs

f4exb, 2021-03-04:
With no limit FPS there is still a limit which is the FFT refresh time that depends on the baseband sample rate, FFT size, averaging and overlap. So firstly I need to know this data to be able to reproduce the case.

Secondly the original 20 FPS capping was there for a good reason to limit the amount of processing in the spectrum display. Depending on you graphics hardware this limit can be pushed to a higher value for a smoother rendering for most of the modern hardware found in the desktop and laptop devices. However with "no limit" you will be able to push this hardware to its actual limits. In these conditions it may break the proper handling of events and lead to the symptoms you describe.

In conclusion this sort of behavior may happen with no limit on the spectrum refresh rate and therefore may not be a valid issue.

Cherenkov11, 2021-03-04:
- My settings are the nexts: FFT window function (Han), FFT size (4k), FFT overlap (0%), Averaging type (Fix), Avg samples (10). Even though I can reproduce the visual glitch with basically any setting.

- I tried it in 2 different PC, both with Archlinux but very different hardware, in my main laptop I have a Ryzen 7 and a RTX 2060, with NL CPU sits on 15% usage and GPU at 35% max. On my old desktop, CPU goes to 65% on CPU and probably the GeForce 8400 SE maxed out. But in both machines 200 fps works very smoothly.

- What it's odd is that this looks more like a glith, because the program never freezes, all demodulator keep working, it's just that after you zoom in/out you need to move the cursor of the mouse out of the spectrum/waterfall screen to defrost it.

I understand that the NL option is probably not needed, since 200 fps works flawlessly.

f4exb, 2021-03-07:
I shall remove NL setting to prevent abuse then. Moreover the FPS limitation has the advantage of giving a chance to the event loop to handle the refresh in all cases.

commit 01170eb7 (2021-03-07): Spectrum: removed FPS 'no limit' setting. Fixes #793
closed 2021-03-07 by f4exb