f4exb/sdrangel #795: SDRangel can hang if File Sink input freq offset changed while recording

srcejon, 2021-03-05:
It seems SDRangel will hang if the File Sink input freq offset is changed while recording.

It seems very reproducible on Windows.

- Add file sink
- Set file name
- Start acquisition
- Start recording
- Wait a few seconds
- Adjust offset
- SDRangel hangs

But doesn't appear to be a problem on Linux.

Tested with SDRangel 6.6.1.

On a related note, it seems a new sdriq file is created everytime the offset is changed, which means you can't record (and then replay) a file with Doppler correction. Could possibly write a utility to join them, but you'd end up with thousands of files.

f4exb, 2021-03-05:
Initially creating a new file makes sense every time the meta data which comprises center frequency changes. While it would be necessary if sample rate changes it is not mandatory for frequency change if you don't mind that the displayed frequency when playing back is incorrect so this is debatable. If you want to record frequency changes then the SigMF format is better although changing frequency constantly has a certain overhead as each frequency segment still creates a separate section in the meta file.

srcejon, 2021-03-06:
Yep, not perfect. Maybe we could have a check box that determines whether files are split or not?

f4exb, 2021-03-07:
I am not against seamless process while the frequency changes. This is a known limitation with this kind of recording that has a single global metadata header. As mentioned if you want some finer process you should turn to SigMF file recording and even though frequency changes are taken into account only of recorded with SDRangel since this is a specific extension of SigMF  standard for SDRangel. Moreover it is fairly complex to implement a switchable option as the file splitting is done in the `FileRecorder` and not within the plugin sources.

f4exb, 2021-03-07:
Implemented in v6.6.2

closed 2021-03-07 by f4exb