f4exb/sdrangel #796: sdrangel segfault

fedya, 2021-03-05:
```
Program received signal SIGILL, Illegal instruction.
0x00007ffff7d807a8 in std::_Deque_base<std::__detail::_StateSeq<std::__cxx11::regex_traits<char> >, std::allocator<std::__detail::_StateSeq<std::__cxx11::regex_traits<char> > > >::_M_initialize_map(unsigned long) () from /usr/lib/sdrangel/libsdrbase.so
```

bt full

```
0x00007ffff7d807a8 in std::_Deque_base<std::__detail::_StateSeq<std::__cxx11::regex_traits<char> >, std::allocator<std::__detail::_StateSeq<st
d::__cxx11::regex_traits<char> > > >::_M_initialize_map(unsigned long) () from /usr/lib/sdrangel/libsdrbase.so
(gdb) bt full
#0  0x00007ffff7d807a8 in std::_Deque_base<std::__detail::_StateSeq<std::__cxx11::regex_traits<char> >, std::allocator<std::__detail::_StateSe
q<std::__cxx11::regex_traits<char> > > >::_M_initialize_map(unsigned long) () from /usr/lib/sdrangel/libsdrbase.so
No symbol table info available.
#1  0x00007ffff7d7e9f8 in std::__detail::_Compiler<std::__cxx11::regex_traits<char> >::_Compiler(char const*, char const*, std::locale const&,
 std::regex_constants::syntax_option_type) () from /usr/lib/sdrangel/libsdrbase.so
No symbol table info available.
#2  0x00007ffff7d7e86a in std::__cxx11::basic_regex<char, std::__cxx11::regex_traits<char> >::basic_regex<char const*>(char const*, char const
*, std::locale, std::regex_constants::syntax_option_type) () from /usr/lib/sdrangel/libsdrbase.so
No symbol table info available.
#3  0x00007ffff7d7e7d5 in std::__cxx11::basic_regex<char, std::__cxx11::regex_traits<char> >::basic_regex<char const*>(char const*, char const
*, std::regex_constants::syntax_option_type) () from /usr/lib/sdrangel/libsdrbase.so
No symbol table info available.
#4  0x00007ffff7da9b3a in _GLOBAL__sub_I_webapiadapterinterface.cpp () from /usr/lib/sdrangel/libsdrbase.so
No symbol table info available.
```

srcejon, 2021-03-05:
Is it reproducible? 

What were you doing at the time?

f4exb, 2021-03-05:
Usually illegal instruction SIGILL is an exception when attempting to execute code not compatible with the CPU. My bet is that this is old hardware with no SSE 4.2 support.

fedya, 2021-03-05:
my cpu is 
`Intel(R) Xeon(R) CPU E3-1220 V2 @ 3.10GHz`
full build.log here https://abf.openmandriva.org/build_lists/1398

f4exb, 2021-03-06:
`Xeon(R) CPU E3-1220`  has it all: https://ark.intel.com/content/www/us/en/ark/products/52269/intel-xeon-processor-e3-1220-8m-cache-3-10-ghz.html

f4exb, 2021-03-08:
In any case this is inevitable if you are using mismatched binaries vs CPU capabilities. A short note in the Wiki may be added as a warning.

fedya, 2021-03-08:
Well, yes, you right, i built it on my pc and it's working fine.
Probably need to disable avx2 on non-avx2 systems

closed 2021-03-08 by fedya