f4exb/sdrangel #802: REST Delete Channel crashes SDRAngel

N5BRG, 2021-03-08:
When I issue a Delete request to delete a channel SDRAngel can crash.

You can reproduce this by opening the SWAGGER interface and execute the command:
curl -X DELETE "http://127.0.0.1:8091/sdrangel/deviceset/0/channel/0" -H "accept: application/json"
After deleting the channel present add a channel back in manually. Then execute the above command a second time. The system crashes on the second try.

I found this while deveopling a rig module I plan to submit to Hamlib for use with SDRAngel.
Goal is to use the CRLOG program which allows user to pick a station from a DX list and this will set the radio at correct Frequency and mode. I have this all working ececpt it crashes on deleting mode. My plan was to delete the current mode and then post the mode required. I plan to post a fork on Hamlib and have user hand delete the unused mode until I can sort this out.

Details on my setup:
Linux BBQRibs 5.8.0-44-generic #50-Ubuntu SMP Tue Feb 9 06:29:41 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
Intel NUC I5   Ubuntu 20.10

SDRangel Version 6.5.2-6-gdc63f8565   Build info Qt 5.14.2 64 bits
Build locally from GitHub.

Problem exist from Hamlib using my code also but SWAGGER verification is probable best for now. I can supply more details if needed.

Portion of Crash data:
021-03-08 07:59:11.957 (D) SpectrumVis::applySettings:  m_fftSize:  1024  m_fftWindow:  4  m_fftOverlap:  0  m_averagingIndex:  0  m_averagingMode:  0  m_refLevel:  0  m_powerRange:  100  m_linear:  false  m_ssb:  true  m_usb:  true  m_wsSpectrumAddress:  "127.0.0.1"  m_wsSpectrumPort:  8887  force:  false
2021-03-08 07:59:11.957 (D) DSPDeviceSinkEngine::addChannelSource:  SSBMod
2021-03-08 07:59:11.958 (D) DSPDeviceSinkEngine::handleSynchronousMessages:  DSPAddBasebandSampleSource
2021-03-08 07:59:11.958 (D) NFMMod::handleMessage: DSPSignalNotification
2021-03-08 07:59:11.958 (D) ValueDialZ::formatText: value: 0
2021-03-08 07:59:11.968 (D) GLSpectrumGUI::on_averaging_currentIndexChanged: log
2021-03-08 07:59:11.968 (D) GLSpectrumGUI::on_fftSize_currentIndexChanged: 3
2021-03-08 07:59:11.969 (D) GLSpectrum::connectTimer
2021-03-08 07:59:11.969 (D) ValueDialZ::formatText: value: 0
2021-03-08 07:59:11.970 (D) SSBModGUI::applyBandwidths:  dsb:  false  spanLog2:  3  m_spectrumRate:  6000  bw:  30  lw:  3  bwMax:  60  tickInterval:  5
2021-03-08 07:59:11.970 (D) GLSpectrumGUI::handleInputMessages: message: GLSpectrum::MsgReportSampleRate
2021-03-08 07:59:11.970 (D) NFMMod::handleMessage: MsgConfigureSSBMod
2021-03-08 07:59:11.970 (D) SpectrumVis::handleMessage: MsgConfigureSpectrumVis
2021-03-08 07:59:11.970 (D) SpectrumVis::applySettings:  m_fftSize:  1024  m_fftWindow:  4  m_fftOverlap:  0  m_averagingIndex:  0  m_averagingMode:  0  m_refLevel:  0  m_powerRange:  100  m_linear:  false  m_ssb:  true  m_usb:  true  m_wsSpectrumAddress:  "127.0.0.1"  m_wsSpectrumPort:  8887  force:  false
2021-03-08 07:59:11.972 (D) GLSpectrum::initializeGL: context:  major:  4  minor:  6  ES:  no
2021-03-08 07:59:11.973 (D) GLShaderTextured::initializeGL: m_useImmutableStorage:  true
2021-03-08 07:59:11.973 (D) GLShaderTextured::initializeGL: m_useImmutableStorage:  true
2021-03-08 07:59:11.973 (D) GLShaderTextured::initializeGL: m_useImmutableStorage:  true
2021-03-08 07:59:11.973 (D) GLShaderTextured::initializeGL: m_useImmutableStorage:  true
2021-03-08 07:59:11.973 (D) GLShaderTextured::initializeGL: m_useImmutableStorage:  true
2021-03-08 07:59:16.140 (D) HttpConnectionHandler (0x56369090b200): read timeout occurred
2021-03-08 07:59:16.141 (D) HttpConnectionHandler (0x56369090b200): disconnected
2021-03-08 07:59:18.509 (D) HttpConnectionHandler (0x56369090b200): received request from 127.0.0.1 (DELETE) /sdrangel/deviceset/1/channel/0
2021-03-08 07:59:18.509 (D) MainWindow::handleMessages: message: MainCore::MsgDeleteChannel
./sdrangel.sh: line 1: 301331 Segmentation fault      /opt/install/sdrangel/bin/sdrangel

If there is a way to work around the issue that would be fine. I tried PATCHing the existing channel but does not work unless you are using the same mode. USB for example.

Bob

f4exb, 2021-03-15:
Sorry I am not able to reproduce with the information you give and latest head of master (commit af98689c3)

N5BRG, 2021-03-16:
Problem seems to be corrected!

I pulled down most current master which builds Version 6.6.2-16-g999e4d781 and built this package. Then I tested for the crash and do not see a proble.

It is nice to see a problem like this just go away while you sleep. :-)

Thanks,

Bob

closed 2021-03-16 by N5BRG