f4exb/sdrangel #805: Debug build fails to link on Windows

srcejon, 2021-03-09:
The debug configuration fails to build on Windows, with a linker error in modatv to functions in OpenCV.

This appears to be because the debug version of the opencv library hasn't been included in the the sdrangel-windows-libraries in \opencv4\x64\vc16\lib\opencv_world450d.lib

I can build with the vc15 version of the library, but wanted to fix with the vc16 version. However, I couldn't see where you got that build from as it doesn't appear to be in the standard opencv windows release.

Any chance you could add it please? Thanks.

srcejon, 2021-03-09:
Ahh, it seems it was @kasper93 that added OpenCV for VS2019 in this pull request:

https://github.com/f4exb/sdrangel-windows-libraries/pull/7

Can this issue be assigned to him? All I think we need is for the missing opencv_world450d.lib file to be added.

f4exb, 2021-03-09:
"arobasing" his user might be enough. Indeed it looks fairly easy to add it here: https://github.com/f4exb/sdrangel-windows-libraries/tree/master/opencv4/x64/vc16/lib but I don't know where it came from originally.

f4exb, 2021-03-09:
I am not able to get the "vc16" version from the usual sources (by Googling). Just vc14 and vc15.

srcejon, 2021-03-09:
Hmm, that's strange - the vc15 version is only 3MB.

f4exb, 2021-09-03:
> but we need to use LFS 

Well if I got it right it is free up to 2 GB so why not... However should not the place rather be this repo: https://github.com/f4exb/sdrangel-windows-libraries ?

srcejon, 2021-09-03:
I think so - missing file should be in here: https://github.com/f4exb/sdrangel-windows-libraries/tree/master/opencv4/x64/vc16/lib

f4exb, 2021-09-03:
Since https://github.com/f4exb/sdrangel-windows-libraries is a fork LFS cannot be installed on it. It has to be "de-forked".

f4exb, 2021-09-03:
So... I have imported sdrangel-windows-libraries into a new repo sdrangel-windows-libraries-lfs and applied LFS tracking for ,dll, .pdb, .exe and .lib files. I have just changed the path to this new repo in external/windows. Let's see if this works ...

f4exb, 2021-09-03:
It looks like it is working. I will keep the old forked repository for some time then delete it to avoid confusion.

f4exb, 2021-11-04:
Space may be free up to 2 (or 1?) GB the download bandwidth is not and this makes this option not an option. In particular it is billed at each actions CI/CD run. So eventually I have returned to the original https://github.com/f4exb/sdrangel-windows-libraries

closed 2022-02-07 by github-actions[bot]