f4exb/sdrangel #806: Crash when changing frequency using a Preset

V96GLF, 2021-03-09:
My implementation of SDRangel crashes intermittently (around 50% of the time) when changing frequency by double-clicking a preset, whilst the sampling device is running.

If I stop the device first, then invoke the preset, then restart the device, then the crash does not happen.

If using the Satellite Tracker feature, then it crashes 100% of the time when an AOS is invoked that requires a frequency change.

SDRangel 6.6.0
Windows 10.0.18363
Nooelec NESDR Smart
Zadig - WinUSB (v6.1.7600.16385)

Sample log file generated for a crash:

`2021-03-09 11:44:55.856 (D) MainWindow::on_presetLoad_clicked
2021-03-09 11:44:55.856 (D) MainWindow::loadPresetSettings: preset [DMR | Pi-Star]
2021-03-09 11:44:55.856 (D) GLSpectrumGUI::applySettings: refLevel: 0.000000e+00 powerRange: 1.000000e+02
2021-03-09 11:44:55.857 (D) SpectrumVis::handleMessage: MsgConfigureSpectrumVis
2021-03-09 11:44:55.857 (D) SpectrumVis::applySettings:  m_fftSize:  1024  m_fftWindow:  4  m_fftOverlap:  0  m_averagingIndex:  0  m_averagingMode:  0  m_refLevel:  0  m_powerRange:  100  m_fpsPeriodMs:  50  m_linear:  false  m_ssb:  false  m_usb:  true  m_wsSpectrumAddress:  "127.0.0.1"  m_wsSpectrumPort:  8887  force:  false
2021-03-09 11:44:55.857 (D) DeviceAPI::loadSamplingDeviceSettings: Loading Rx preset [DMR | Pi-Star]
2021-03-09 11:44:55.857 (D) Preset::findBestDeviceConfig: serial matched (exact): id: sdrangel.samplesource.rtlsdr ser: 00000001
2021-03-09 11:44:55.857 (D) DeviceAPI::loadSamplingDeviceSettings: source center frequency: 438750000 Hz
2021-03-09 11:44:55.857 (D) DeviceAPI::loadSamplingDeviceSettings: deserializing source sdrangel.samplesource.rtlsdr[0]: 00000001
2021-03-09 11:44:55.857 (D) RTLSDRInput::handleMessage: MsgConfigureRTLSDR
2021-03-09 11:44:55.864 (C) RTLSDRInput::applySettings: could not set AGC mode off
2021-03-09 11:44:55.864 (D) DSPDeviceSourceEngine::configureCorrections
2021-03-09 11:44:55.864 (D) RTLSDRInput::applySettings: corrections: DC block: true IQ imbalance: false
2021-03-09 11:44:55.864 (C) RTLSDRInput::applySettings: could not set LO ppm correction: 0
2021-03-09 11:44:55.864 (D) DSPDeviceSourceEngine::handleInputMessages: message: DSPConfigureCorrection
2021-03-09 11:44:56.072 (D) RTLSDRInput::applySettings: sample rate set to 1024000
2021-03-09 11:44:56.072 (D) RTLSDRInput::applySettings: log2decim set to 2
2021-03-09 11:44:56.072 (D) RTLSDRInput::applySettings: set fc pos (enum) to  2
2021-03-09 11:44:56.072 (D) DeviceSampleStatic::calculateSourceDeviceCenterFrequency:  frequencyShiftScheme:  0  desired center freq:  430680000  Hz  device center freq:  430680000  Hz  device sample rate:  1024000 S/s  Actual sample rate:  256000 S/s  center freq position code:  2  image frequency:  430680000 Hz
2021-03-09 11:44:56.168 (D) RTLSDRInput::applySettings: rtlsdr_set_center_freq(430680000)
2021-03-09 11:44:56.168 (D) RTLSDRInput::applySettings: set noModMode to  false
2021-03-09 11:44:56.708 (D) RTLSDRInput::applySettings: set RF bandwidth to  2500000
2021-03-09 11:44:56.708 (C) RTLSDRInput::applySettings: could not set offset tuning to off`

(I have searched, but haven't found this reported elsewhere)

srcejon, 2021-03-09:
Fancy seeing you here :)

Do you happen to have the ChannelAnalyzer open? 

I've just tried to reproduce this - and I can in one instance when the Channel Analyzer is open. I'm getting an exception in ChannelAnalyzerSink::feed line 94. Looks like m_sampleSink is invalid (Not null, but pointing to something I presume has been deleted).

If you don't have the ChannelAnalyzer open, it would be useful to know what channels you do have open. Perhaps try creating a preset with no channels/features open, and see if it still crashes (that seems to be fine for me).

V96GLF, 2021-03-10:
No, I don't have ChannelAnalyzer open.

Good idea regarding the presets with no channels/features open. I have created three, on different frequencies, and I can jump between them with no problem.

If I open a presets that has DSD demodulator (only), and then double-click any of the test (empty) presets, it crashes every time (unless I stop the device first).

srcejon, 2021-03-10:
Ok, that is reproducible too. Thanks.

I see an exception in void DSDDemodSink::feed line 302 - as with the channel analyzer, the m_scopeXY pointer is pointing to a deleted object. 

Presumably can be fixed by having the preset loader pause acquisition, but probably best for @f4exb to take a look.

f4exb, 2021-03-10:
Thanks for having a closer look. It seems `m_scopeXY` life cycle is not handled properly as it should either point to a valid object or be `nullptr` and then checked before use. I'll check that.

f4exb, 2021-03-15:
Same problem with
  - Channel Analyzer (already spotted)
  - DATV Demod

commit 43059e39 (2021-03-15): Set depending objects to nullptr before deleting GUI. Fixes #806
closed 2021-03-15 by f4exb