f4exb/sdrangel #808: Satellite Tracker: can't  set "Preset to load on AOS"

jh4xsy, 2021-03-16:
I want to decode NOAA APT image.
But I can't set "137.500 - APT" at "Preset to load on AOS" at SDRangel Control dialog.
 
Followed readme.md:

1. Select "NOAA 18" from the dropdown list. then Information about the satellite's transmit and receive modes appeared in the field at the bottom of the dialog.
2. Press the + button.
3. I can select "Device set" from dropdown list.
4. No dropdown list at "The preset to load on AOS" 

I use v6.6.2 windows binary and compiled from source at linux-box(openSUSE 15.2).
and APT Demodulator Settings: Enable Satellite Tracker control is checked.

[image]

srcejon, 2021-03-16:
The preset does not exist as standard, you need to create it yourself. Have you created the preset?

i.e. setup SDRangel how you want it, with the APT demod open and frequency set. Then press the "Save current settings as new preset" button in the Presets tab.

jh4xsy, 2021-03-16:
I thought that dropdown lists are generated from SatNOGS info like gpredict. But it use Presets! 

After I saved Presets, it works. Thanks, srcejon.

closed 2021-03-16 by jh4xsy