f4exb/sdrangel #835: Crash with only tx device

hiperiondev, 2021-04-04:
I have a soapysdr driver with only tx channel.
SoapySDRUtil --probe respond:
----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 0 Rx, 1 Tx
  Timestamps: NO

----------------------------------------------------
-- TX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CF32
  Native format: CF32 [full-scale=1]
  Full gain range: [0, 0] dB
  Full freq range: [0.05, 600] MHz
    RF freq range: [0.05, 600] MHz
  Sample rates: 0.02, 0.05, 0.1, 0.25, 0.5, 1.25 MSps

if I add a sink with this device crash with:
/usr/include/c++/10/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = DeviceSoapySDRParams::ChannelSettings; _Alloc = std::allocator<DeviceSoapySDRParams::ChannelSettings>; std::vector<_Tp, _Alloc>::reference = DeviceSoapySDRParams::ChannelSettings&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.

If I define an RX channel Sdrangel works as expected.l

f4exb, 2021-04-05:
Firstly you cannot have no RX0 in the GUI version. Shall I conclude that you are dealing with the server flavor? Please be more specific in your description.

Secondly I do not recommend using Soapy SDR. Anyway it is not in my plans to give any kind of support or follow-up for Soapy SDR related issues.

hiperiondev, 2021-04-05:
I am implementing a generic driver for rpitx (on raspberry pi zero), which is just a transmitter.
As a receiver I use an rtl-sdr. 
Do you have any recommendations on another type of driver that could be used instead of SoapySDR and would be more convenient? 
rpitx is an excellent option to transmit at a very low cost

f4exb, 2021-04-05:
Well ideally one would write a plugin for RpiTx. This ticket is still waiting for implementation: https://github.com/f4exb/sdrangel/issues/216

f4exb, 2021-04-06:
In any case the remote output device and remote output channel sink design should be revised. There is a feedback to adjust the exact stream sample rate to compensate for clock skews. In fact the remote should adapt by itself through occasional decimation or interpolation of samples and thus be completely autonomous. This could be the object of a new ticket as not being obviously related.

f4exb, 2021-04-10:
Closed in favor of #216 and #838

closed 2021-04-10 by f4exb