f4exb/sdrangel #840: Doppler correction in Satellite Tracker Feature

Valstee, 2021-04-12:
Hi F4exb,
I use SDRangel 6.8.0-11-g30c2c519d Qt 5.14.2 x86_64 on Ubuntu 20.10.
In Satellite tracker feature, I can't check a channel to enable Doppler correction:

[image]

An idea to solve this issue ?
Thanks,
Valentin (from Antibes)

f4exb, 2021-04-12:
What is the issue?

srcejon, 2021-04-12:
The check boxes next to the channel names are missing - which should allow the doppler correction to be enabled/disabled per channel.

They appear on Windows with qt 5.15 for me, but are missing on Ubuntu with Qt 5.14 for me also, so I'll see if it can be done differently.

f4exb, 2021-04-12:
I see... same thing for me on Ubuntu 20.10 and Qt 5.14.2

Valstee, 2021-04-14:
Thanks a lot for your response.
Same thing for me : They appear on Windows with qt 5.15, but are missing on Ubuntu with Qt 5.14 (that's how I saw the "issue")

srcejon, 2021-04-17:
Works in kubuntu 21.04 which has Qt 5.15.2 - so presumably an issue with Qt 5.14. Can't find a work around. Perhaps this dialog would be clearer if it used tabs (one per device set) with a more standard form layout, rather than a table anyway.

f4exb, 2021-04-18:
We still need compatibility down to 5.12.8 for the Ubuntu 20.04 package. I recently upgraded the base image of the Docker build to Debian bullseye therefore lifting the 5.11 compatibility.

srcejon, 2021-07-04:
With the above path, I've rearranged the GUI to use tabs rather than a tab. Now works for me on Ubuntu with Qt 5.14 and Windows.

Valstee, 2021-07-06:
Thanks a lot for the modifications !

closed 2021-07-06 by Valstee