f4exb/sdrangel #844: Support for streaming with the rtmp and srt protocols to the datv modulator

computer-wizard, 2021-04-14:
Request for support of the RTMP and SRT  (secure reliable transport) streaming protocols as this gives a much better user
experience as UDP.   RTMP and SRT are both supported by OBS-studio and FFMPEG

f4exb, 2021-04-14:
For RTMP you may as well use nginx on the host with nginx rtmp module installed. In the nginx.conf you would code a block like that:
```
rtmp {
       server {
           listen 192.168.0.3:1935;
           chunk_size 4096;
           
           application sdrangel {
               live on;

               # captation flux
               exec ffmpeg -i rtmp://192.168.0.3:1935/sdrangel/$name -codec copy -f mpegts -metadata service_provider=SDRangel -metadata service_name=$name -streamid 0:256 udp://127.0.0.1:8872;
           } 
       }
}
```
It reads the RTMP stream on `192.168.0.3:1935` and translates it to UDP using `ffmpeg` directing the UDP stream to `127.0.0.1:8872` 

While it could be more convenient to have RTMP natively implemented in the plugin this could serve as a workaround for RTMP support.

computer-wizard, 2021-04-15:
Thanks for info

f4exb, 2021-06-23:
will do with the workaround suggested

closed 2021-06-23 by f4exb