f4exb/sdrangel #845: Segmentation error with satellite tracker when AOS

Valstee, 2021-04-14:
Hi @f4exb , Hi @srcejon,
I use SDRangel 6.9.0, QT 5.14.2 on Ubuntu 20.10 and satellite tracker feature.
When there is a change of satellite (and therefore a change of preset) during an acquisition, it causes a segmentation error.
(this change can be done by an AOS of a priority satellite or manually)
Valentin

`
2021-04-14 16:22:14.067 (D) RTLSDRInput::handleMessage: MsgConfigureRTLSDR
2021-04-14 16:22:14.068 (D) RTLSDRInput::applySettings: log2decim set to 2
2021-04-14 16:22:14.068 (D) DeviceSampleStatic::calculateSourceDeviceCenterFrequency:  frequencyShiftScheme:  0  desired center freq:  95800000  Hz  device center freq:  95800000  Hz  device sample rate:  1024000 S/s  Actual sample rate:  256000 S/s  center freq position code:  2  image frequency:  95800000 Hz
2021-04-14 16:22:14.122 (D) RTLSDRInput::applySettings: rtlsdr_set_center_freq(95800000)
2021-04-14 16:22:14.122 (D) DSPDeviceSourceEngine::handleInputMessages: message: DSPSignalNotification
2021-04-14 16:22:14.122 (D) DSPDeviceSourceEngine::handleInputMessages: DSPSignalNotification:  m_sampleRate:  256000  m_centerFrequency:  95800000
2021-04-14 16:22:14.122 (D) DSPDeviceSourceEngine::handleInputMessages: forward message to  SpectrumVis
2021-04-14 16:22:14.122 (D) SpectrumVis::handleMessage: DSPSignalNotification:  centerFrequency:  95800000  sampleRate:  256000
2021-04-14 16:22:14.122 (D) DSPDeviceSourceEngine::handleInputMessages: forward message to  FileSink
2021-04-14 16:22:14.122 (D) FileSink::handleMessage: DSPSignalNotification:  inputSampleRate:  256000  centerFrequency:  95800000
2021-04-14 16:22:14.123 (D) DSPDeviceSourceEngine::handleInputMessages: forward message to  WFMDemod
2021-04-14 16:22:14.123 (D) WFMDemod::handleMessage: DSPSignalNotification
2021-04-14 16:22:14.123 (D) FileSinkBaseband::handleMessage: DSPSignalNotification:  basebandSampleRate:  256000  cnterFrequency:  95800000
2021-04-14 16:22:14.123 (D) DSPDeviceSourceEngine::handleInputMessages: DSPSignalNotification: guiMessageQueue: 0x55f5292ca2b0
2021-04-14 16:22:14.123 (D) WFMDemodBaseband::handleMessage: DSPSignalNotification: basebandSampleRate:  256000
2021-04-14 16:22:14.123 (D) DownChannelizer::applyChannelization done:  nb stages: 0  in (baseband): 256000  req: 270000  out (channel): 256000  fc: 0
2021-04-14 16:22:14.123 (D) WFMDemodSink::applyChannelSettings:  channelSampleRate:  256000  channelFrequencyOffset:  0
2021-04-14 16:22:14.124 (D) NCO freq: 0.000000 phase inc 0
2021-04-14 16:22:14.124 (D) WFMDemod::applyChannelSettings: m_interpolator.create
2021-04-14 16:22:14.124 (D) Interpolator::createPolyphaseLowPass: ntaps: 72
2021-04-14 16:22:14.124 (D) DownChannelizer::applyChannelization done:  nb stages: 0  in (baseband): 256000  req: 1024000  out (channel): 256000  fc: 0
2021-04-14 16:22:14.124 (D) DownChannelizer::applyChannelization done:  nb stages: 0  in (baseband): 256000  req: 256000  out (channel): 256000  fc: 0
2021-04-14 16:22:14.124 (D) WFMDemod::applySettings: m_rfFilter->create_filter
2021-04-14 16:22:14.124 (D) FileSinkSink::applyChannelSettings:  channelSampleRate:  256000  sinkSampleRate:  256000  channelFrequencyOffset:  0  centerFrequency:  95800000  force:  false
2021-04-14 16:22:14.124 (D) NCOF::setFreq: freq: 0.000000 m_phaseIncrement: 0.000000
2021-04-14 16:22:14.124 (D) FileSinkSink::applyChannelSettings: log2decim:  0
2021-04-14 16:22:14.124 (D) WFMDemod::applySettings: m_fmExcursion: 75000.000000
2021-04-14 16:22:14.125 (D) WFMDemodSink::applyAudioSampleRate: 48000
2021-04-14 16:22:14.125 (D) Interpolator::createPolyphaseLowPass: ntaps: 72
2021-04-14 16:22:14.127 (D) GLSpectrumGUI::handleInputMessages: message: GLSpectrum::MsgReportSampleRate
2021-04-14 16:22:14.127 (D) RTLSDRGui::handleInputMessages: message: DSPSignalNotification
2021-04-14 16:22:14.127 (D) RTLSDRGui::handleInputMessages: DSPSignalNotification: SampleRate:256000, CenterFrequency:95800000
2021-04-14 16:22:14.127 (D) GLSpectrumGUI::handleInputMessages: message: GLSpectrum::MsgReportSampleRate
2021-04-14 16:22:14.128 (D) FileRecord::handleMessage: DSPSignalNotification: inputSampleRate:  256000  centerFrequency:  95800000
2021-04-14 16:22:14.128 (D) FileRecord::stopRecording
2021-04-14 16:22:14.128 (D) FileRecord::startRecording
2021-04-14 16:22:14.128 (D) SpectrumVis::handleMessage: DSPSignalNotification:  centerFrequency:  95800000  sampleRate:  256000
2021-04-14 16:24:23.180 (D) SatelliteTrackerWorker::aos  "AAUSAT-II"
2021-04-14 16:24:23.182 (D) SatelliteTrackerWorker::aos - autoTarget setting  "AAUSAT-II"
2021-04-14 16:24:23.182 (D) SatelliteTracker::handleMessage: MsgConfigureSatelliteTracker
2021-04-14 16:24:23.182 (D) SatelliteTracker::applySettings:  m_latitude:  43.6679  m_longitude:  6.91233  m_heightAboveSeaLevel:  500  m_target:  "AAUSAT-II"  m_satellites:  ("UVSQ-SAT", "ChallengeOne", "AmicalSat", "ISS", "AAUSAT 4", "AAUSAT-II", "CUTE-1.7+APD II", "NOAA 15", "NOAA 18", "NOAA 19", "Bobcat-1", "UWE-3", "CUBESAT XI-IV", "AISAT", "AISAT-1", "Centauri-1", "BRITE-PL", "AIST-2D", "BUGSAT-1", "BEESAT-10", "FIRST-MOVE", "BY70-2", "CAS-2T", "Da Vinci", "al-Farabi-2", "CARTOSAT-2B", "CP11 (ISX)", "CP7 (DAVE)", "FENGYUN 3A", "AMGU-1", "AOBA VELOX-IV", "GOES 15", "ERS-15", "APSS-1", "ARMADILLO", "FUNCUBE-1")  m_tles:  ("https://db.satnogs.org/api/tle/", "https://www.amsat.org/tle/current/nasabare.txt", "https://www.celestrak.com/NORAD/elements/goes.txt")  m_dateTime:  ""  m_minAOSElevation:  0  m_minPassElevation:  0  m_azElUnits:  1  m_groundTrackPoints:  100  m_dateFormat:  "yyyy/MM/dd"  m_utc:  false  m_updatePeriod:  10  m_dopplerPeriod:  1  m_defaultFrequency:  1e+08  m_drawOnMap:  true  m_autoTarget:  true  m_aosSpeech:  "AOS"  m_losSpeech:  "${name} is no longer visible."  m_aosCommand:  ""  m_losCommand:  ""  m_predictionPeriod:  5  m_passStartTime:  QTime("00:00:00.000")  m_passFinishTime:  QTime("23:59:59.000")  m_deviceSettings:  QHash(("BY70-2", 0x55f529a6e7a0)("ERS-15", 0x55f529a6feb0)("NOAA 18", 0x55f529a69d80)("CARTOSAT-2B", 0x55f529a6cba0)("NOAA 15", 0x55f529a6c660)("ARMADILLO", 0x55f529a6ed20)("AISAT-1", 0x55f529a6ce40)("FIRST-MOVE", 0x55f529a6d620)("Centauri-1", 0x55f529a6e300)("AOBA VELOX-IV", 0x55f529a6f200)("FUNCUBE-1", 0x55f529a6d380)("AAUSAT 4", 0x55f529a675b0)("AISAT", 0x55f529a6c900)("ChallengeOne", 0x55f529a6db60)("GOES 15", 0x55f529a68280)("UVSQ-SAT", 0x55f529a6b5b0)("AAUSAT-II", 0x55f529a690b0)("AmicalSat", 0x55f529a6c3c0)("UWE-3", 0x55f529a69490)("BRITE-PL", 0x55f529a68e10)("CP11 (ISX)", 0x55f529a6e550)("al-Farabi-2", 0x55f529a6a2c0)("AIST-2D", 0x55f529a6b310)("Da Vinci", 0x55f529a6bd40)("BEESAT-10", 0x55f529a6fc60)("NOAA 19", 0x55f529a6a020)("FENGYUN 3A", 0x55f529a6de00)("CUTE-1.7+APD II", 0x55f529a67fe0)("ISS", 0x55f529a6f4c0)("CUBESAT XI-IV", 0x55f529a6bfe0)("BUGSAT-1", 0x55f529a6f780)("AMGU-1", 0x55f529a6d8c0)("CP7 (DAVE)", 0x55f529a6d0e0)("CAS-2T", 0x55f529a699a0)("APSS-1", 0x55f529a6ea60)("Bobcat-1", 0x55f529a71ac0))  m_title:  "Satellite Tracker"  m_rgbColor:  4292942179  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIFeatureSetIndex:  0  m_reverseAPIFeatureIndex:  0  force:  false
2021-04-14 16:24:23.182 (D) SatelliteTrackerWorker::applySettings:  m_target:  "AAUSAT-II"  m_satellites:  ("UVSQ-SAT", "ChallengeOne", "AmicalSat", "ISS", "AAUSAT 4", "AAUSAT-II", "CUTE-1.7+APD II", "NOAA 15", "NOAA 18", "NOAA 19", "Bobcat-1", "UWE-3", "CUBESAT XI-IV", "AISAT", "AISAT-1", "Centauri-1", "BRITE-PL", "AIST-2D", "BUGSAT-1", "BEESAT-10", "FIRST-MOVE", "BY70-2", "CAS-2T", "Da Vinci", "al-Farabi-2", "CARTOSAT-2B", "CP11 (ISX)", "CP7 (DAVE)", "FENGYUN 3A", "AMGU-1", "AOBA VELOX-IV", "GOES 15", "ERS-15", "APSS-1", "ARMADILLO", "FUNCUBE-1")  m_dateTime:  ""  m_utc:  false  m_updatePeriod:  10  force:  false
2021-04-14 16:24:23.183 (D) SatelliteTrackerWorker::aos: Loading preset  "Agora-FM"  to  'R'
2021-04-14 16:24:23.187 (D) MainWindow::handleMessages: message: MainCore::MsgLoadPreset
2021-04-14 16:24:23.187 (D) MainWindow::loadPresetSettings: preset [Radio | Agora-FM]
2021-04-14 16:24:23.187 (D) GLSpectrumGUI::applySettings: refLevel: 0.000000e+00 powerRange: 1.000000e+02
2021-04-14 16:24:23.187 (D) SpectrumVis::handleMessage: MsgConfigureSpectrumVis
2021-04-14 16:24:23.187 (D) SpectrumVis::applySettings:  m_fftSize:  1024  m_fftWindow:  4  m_fftOverlap:  0  m_averagingIndex:  0  m_averagingMode:  0  m_refLevel:  0  m_powerRange:  100  m_fpsPeriodMs:  50  m_linear:  false  m_ssb:  false  m_usb:  true  m_wsSpectrumAddress:   [... cut ...]
2021-04-14 16:24:23.758 (W) QOpenGLTexturePrivate::destroy() called but texture context 0x7f546c011880 is not shared with current context 0x7f546c005b40.
Texture has not been destroyed
2021-04-14 16:24:23.759 (W) QOpenGLTexturePrivate::destroy() called but texture context 0x7f546c011880 is not shared with current context 0x7f546c005b40.
Texture has not been destroyed
2021-04-14 16:24:23.759 (W) QOpenGLTexturePrivate::destroy() called but texture context 0x7f546c011880 is not shared with current context 0x7f546c005b40.
Texture has not been destroyed
Segmentation fault (core dumped)
`

srcejon, 2021-04-17:
Hi Valentin,

Does SDRangel crash if you manually switch between the presets while acquisition is running? (i.e. just double click on the preset names back and forth in the main SDRangel GUI)

If not, could you please upload some screenshots showing which plugins you have in each preset?

Also, to help narrow down the source of the crash, could you run SDRangel via gdb?

    sudo apt-get install gdb
    gdb /opt/install/sdrangel/bin/sdrangel
    run

Then after it crashes, type:

    info stack

And then copy and paste the output here. Thanks.

Valstee, 2021-04-19:
Hi Srcejon,
It crashes when I manually change the preset, this happens 1 time out of 2.
The open channels are only filesink and a WFM demodulator.

Here's the output :
`Thread 38 "QThread" received signal SIGSEGV, Segmentation fault. `
`[Switching to Thread 0x7fff7b7fe640 (LWP 69196)]`
`0x00007ffff68d35f4 in QImage::height() const () from /lib/x86_64-linux-gnu/libQt5Gui.so.5`

In fact, this error prevents me from using satellite tracker because when there is a change of satellite and therefore a change of preset, sdrangel crashes...

srcejon, 2021-04-19:
Thanks. Possibly similar to #806

Valstee, 2021-04-19:
Yes, that's exactly what it looks like! Sorry, didn't see this issue...

srcejon, 2021-04-19:
No problem, this looks like an additional problem - and I can reproduce it - looks to be from the FileSink, from the full trace:

(gdb) info stack
#0  0x00007ffff675d5f4 in QImage::height() const () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#1  0x00007ffff7e3887e in GLSpectrum::updateWaterfall(float const*) (this=0x555556c5d900, spectrum=0x555556c43b60)
    at /opt/build/srcejon/sdrangel/sdrgui/gui/glspectrum.cpp:447
#2  0x00007ffff7e387de in GLSpectrum::newSpectrum(float const*, int, int)
    (this=0x555556c5d900, spectrum=0x555556c43b60, nbBins=1024, fftSize=1024) at /opt/build/srcejon/sdrangel/sdrgui/gui/glspectrum.cpp:441
#3  0x00007ffff74a007d in SpectrumVis::feed(__gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, __gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, bool) (this=0x555555908668, cbegin=
  {m_real = -54528, m_imag = 206592}, end={m_real = -246208, m_imag = -184000}, positiveOnly=false)
    at /opt/build/srcejon/sdrangel/sdrbase/dsp/spectrumvis.cpp:410
#4  0x00007fffae225690 in FileSinkSink::feed(__gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, __gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&) (this=0x555556c595a0, begin=
  {m_real = -54528, m_imag = 206592}, end={m_real = -246208, m_imag = -184000})
    at /opt/build/srcejon/sdrangel/plugins/channelrx/filesink/filesinksink.cpp:191
#5  0x00007ffff744ae85 in DownChannelizer::feed(__gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, __gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&) (this=0x555556fe2620, begin=
  {m_real = -54528, m_imag = 206592}, end={m_real = -246208, m_imag = -184000}) at /opt/build/srcejon/sdrangel/sdrbase/dsp/downchannelizer.cpp:55
#6  0x00007fffae2235e4 in FileSinkBaseband::handleData() (this=0x555556c59530)
    at /opt/build/srcejon/sdrangel/plugins/channelrx/filesink/filesinkbaseband.cpp:108

Valstee, 2021-04-19:
Thanks for these precisions. I confirm that if I remove Filesink from my presets, there is no crash.

PS: In the satellite tracker feature, is it necessary to have filesink open for an automatic record of a satellite pass ?

srcejon, 2021-04-19:
If you want to record the IQ data from a pass to a file, then yes, you need the filesink.

f4exb, 2021-04-19:
I suppose that your setup is to record satellite passes in unattended mode. Then the server variant (no GUI) could be used (also on Raspberry Pi) and since there is no GUI there is no GLSpectrum either so that probably works around the issue. I would like to put up a similar setup with a RaspPi sometime... Maybe a few things would be to be ironed out but in principle this should work.

Anyway this does not solve the issue. It looks like a race condition but as far as I could see in the code it seems mutexes are used everywhere.

Valstee, 2021-04-19:
Thanks for your answers ! I just bought a raspberry pi4 last week and SDR angel is already running on it. I'm going to test the server version. I'll keep you informed.

f4exb, 2021-04-19:
I am trying to load a preset with a FileSink while the device is running but I cannot reproduce the issue. Is there anything more special?

srcejon, 2021-04-19:
I don't think so. Have tried using a RTL SDR and USRP. Had two presets with just a file sink and nothing else. Press run then switch between the two a few times. Happens on both windows and linux.

Valstee, 2021-04-19:
In my case, the frequency of the presets is different...
This happens with linux and window with my RTL SDR and my limeSDR.

V96GLF, 2021-07-21:
I'm still seeing this issue when using Satellite Tracker.

If Satellite Tracker invokes a change of preset (or change of frequency with the same preset), then SDRangel crashes.

Let me know if any additional details would be helpful.

srcejon, 2021-07-21:
Do you have the File Sink in your preset? If not, which channels do you have?

V96GLF, 2021-07-21:
No File Sink.

I'm usually running APT Demodulator, NFM Demodulator, and Packet Demodulator - but I haven't experimented with removing any of them. Could try that?

srcejon, 2021-07-21:
Yes please - it may help narrow down in which channel is the problem. 

Also, does it crash only when Sat Tracker changes the preset - or if you manually switch between two presets by double clicking in the presets tab?

(I've just tried the latter 50+ times without an issue for various combinations of those channels - although it seems I no longer can reproduce the problem with the File Sink either).

V96GLF, 2021-07-22:
It now seems stable if I only have APT Demodulator open. I think the crash is associated with changing frequency whilst the receiver is running - but I'll continue to test and see if I can come up with anything reproducable.

It currently seems stable when double-clicking between presets.

V96GLF, 2021-07-27:
Further testing...

I have 5 satellites configured (NO-84, NOAA 15, 18, 19, ISS) as presets, each with only one Channel: APT Demodulator (or Packet Demodulator for ISS).

Satellite Tracker has each satellite mapped to a preset, with Doppler correction, AOS and LOS checked; start/stop filesink unchecked.

Starting with the receiver not running the **first** satellite pass works fine. AOS is announced, image is received correctly, LOS is announced and receiver returns to idle.

Following LOS, it is now "primed" for a crash, which might (but not always) be triggered by either of:
1. The next AOS, or
2. Selecting another preset

If a second satellite overlaps the first, then it transitions to the second satellite, selects the second preset, and announces AOS. But it then crashes at LOS of the second satellite.

srcejon, 2021-07-28:
I've tried to recreate this on Windows using a debug build with an RTL SDR. While I'm not sure I have recreated it, I've seen a couple of exceptions within libusb that is called by the rtlsdr driver:

[image]

SDRangel on Windows is currently released with libusb 1.0.23.11397. 

Trying with the latest, which is 1.0.24.11584 still results in exceptions:

[image]

[image]

srcejon, 2021-11-15:
@V96GLF - I've tried to reproduce this using some dummy satellite data that has satellite passes every minute, but can't seem to get it to crash.

Do you still have the Ubuntu VM set up? We can try to see what's going on there. You can try:

    sudo apt-get install gdb
    gdb /opt/install/sdrangel/bin/sdrangel
    run

Then after it crashes, type:

    info stack

And post the output, so we can see where it's actually crashing.

V96GLF, 2021-11-16:
Yes, I’m still running as a Ubuntu VM.

You know that thing when you make a doctor’s appointment, and the symptoms magically vanish? That. ;-)

I can’t currently get it to crash whilst I’m watching due to Sat Tracker. I do however get the occasional crash when accessing menus (e.g. clicking Window/Features menu).

I’ll continue testing Sat Tracker for a few more days…

srcejon, 2021-11-16:
Heisenbugs, as they're known :)

RE window/features menu crash - running with gdb as above should help track that down. A screenshot beforehand would also be useful. Thanks!

V96GLF, 2021-11-19:
I did mention that I'm running on a different (and "clean") PC, didn't I? Well I think at least some of my problems have been hardware-related:
a) The Nooelec dongle was a little loose in the USB socket, especially with coax hanging off the back of it. I suspect it might have been disconnecting occasionally.
b) My Logitech wireless mouse was behaving erratically. I don't think the USB dongle was happy on the rear of the PC. Perhaps some random mouse clicks as a result?
Both now resolved by adding a USB hub.

Overall much more stable now, including Sat Tracker :-)

However, I do still get occasional crashes when double-clicking a preset whilst acquisition is running (as in the original title of the thread). Here's an example.

Running my "S20-S24" preset, and then double-click the Sat Tracker preset.

[image]

`
Thread 1 "sdrangel" received signal SIGSEGV, Segmentation fault.
0x00007fff9e40e636 in QSGNode::removeAllChildNodes() () from /lib/x86_64-linux-gnu/libQt5Quick.so.5

(gdb) info stack
#0  0x00007fff9e40e636 in QSGNode::removeAllChildNodes() () at /lib/x86_64-linux-gnu/libQt5Quick.so.5
#1  0x00007fff8d2a04b3 in QGeoMapObjectQSGSupport::updateMapObjects(QSGNode*, QQuickWindow*) () at /lib/x86_64-linux-gnu/libQt5Location.so.5
#2  0x00007fff8d297cda in QGeoTiledMapLabs::updateSceneGraph(QSGNode*, QQuickWindow*) () at /lib/x86_64-linux-gnu/libQt5Location.so.5
#3  0x00007fff8d226733 in QDeclarativeGeoMap::updatePaintNode(QSGNode*, QQuickItem::UpdatePaintNodeData*) ()
    at /lib/x86_64-linux-gnu/libQt5Location.so.5
#4  0x00007fff9e4e66b8 in QQuickWindowPrivate::updateDirtyNode(QQuickItem*) () at /lib/x86_64-linux-gnu/libQt5Quick.so.5
#5  0x00007fff9e4e6c63 in QQuickWindowPrivate::updateDirtyNodes() () at /lib/x86_64-linux-gnu/libQt5Quick.so.5
#6  0x00007fff9e4e88a0 in QQuickWindowPrivate::syncSceneGraph() () at /lib/x86_64-linux-gnu/libQt5Quick.so.5
#7  0x00007fff9e57639d in QQuickRenderControl::sync() () at /lib/x86_64-linux-gnu/libQt5Quick.so.5
#8  0x00007fffe8012055 in  () at /lib/x86_64-linux-gnu/libQt5QuickWidgets.so.5
#9  0x00007ffff6f54bfc in QWidget::event(QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff6f116b3 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff642316a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007ffff6f4ad11 in QWidgetPrivate::setGeometry_sys(int, int, int, int, bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007ffff6f4bc10 in QWidget::setGeometry(QRect const&) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007ffff6f34fca in QWidgetItem::setGeometry(QRect const&) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007ffff6f2aa45 in QBoxLayout::setGeometry(QRect const&) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff6f31228 in QLayoutPrivate::doResize() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff6f11697 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff642316a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007ffff6f4ad11 in QWidgetPrivate::setGeometry_sys(int, int, int, int, bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#20 0x00007ffff6f4b372 in QWidget::resize(QSize const&) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007ffff7ee232e in QWidget::resize(int, int)
    (h=<optimised out>, w=<optimised out>, this=0x555559845630, this=<optimised out>, w=<optimised out>, h=<optimised out>)
    at /usr/include/x86_64-linux-gnu/qt5/QtWidgets/qwidget.h:854
#22 RollupWidget::arrangeRollups() (this=0x555555bffc60) at /opt/build/sdrangel/sdrgui/gui/rollupwidget.cpp:240
#23 0x00007ffff7ee245c in RollupWidget::resizeEvent(QResizeEvent*) (this=0x555555bffc60, size=0x7fffffffb600)
    at /opt/build/sdrangel/sdrgui/gui/rollupwidget.cpp:427
#24 0x00007ffff6f54bfc in QWidget::event(QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007ffff6f116b3 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#26 0x00007ffff642316a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#27 0x00007ffff6f4ad11 in QWidgetPrivate::setGeometry_sys(int, int, int, int, bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#28 0x00007ffff6f4bc10 in QWidget::setGeometry(QRect const&) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#29 0x00007ffff6f34fca in QWidgetItem::setGeometry(QRect const&) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#30 0x00007ffff6f2a987 in QBoxLayout::setGeometry(QRect const&) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#31 0x00007ffff6f31228 in QLayoutPrivate::doResize() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#32 0x00007ffff6f32855 in QLayout::activate() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
--Type <RET> for more, q to quit, c to continue without paging--c
#33 0x00007ffff6f11697 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#34 0x00007ffff642316a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#35 0x00007ffff6f4ad11 in QWidgetPrivate::setGeometry_sys(int, int, int, int, bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
[... cut ...]
#98 0x00007ffff6421a9b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#99 0x00007ffff642a024 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#100 0x000055555555708a in runQtApplication(int, char**, qtwebapp::LoggerWithFile*) (argc=<optimised out>, argv=<optimised out>, logger=0x555555588300) at /opt/build/sdrangel/app/main.cpp:130
#101 0x00005555555566cb in main(int, char**) (argc=1, argv=0x7fffffffdfb8) at /opt/build/sdrangel/app/main.cpp:137
(gdb) Quit
`

srcejon, 2021-11-19:
Thanks. One question, the stack trace includes QDeclarativeGeoMap which is related to the Maps (Either in Map Feature or ADS-B). However, I can't see either of those in your screenshot.  

If you select Window > Features, is the Map there?

V96GLF, 2021-11-19:
Yes, it is. No particular reason for having it open, but it is there.

srcejon, 2021-11-19:
Well, that particular crash is due to to something to do with the Map - so if you don't need it, perhaps worth closing it to prevent it causing any problems for you.

I've tried to recreate it, but no problems so far. However, I'm on Ubuntu 20.10 with Qt 5.14.2 - I will install 21.10 with Qt 15.2 to see if that makes any difference (The crash appears to be in Qt code).

V96GLF, 2021-11-19:
Okay, so there is something odd about presets and features... or I'm not driving it correctly.

I go to my S20-24 preset, and close the Map and Sat Tracker features (both were open). I press the "Update selected preset", and "Save the current settings" buttons.

Now I go to the Sat Tracker preset... and the necessary Features (Sat Tracker and Map) are no longer there. So I add them back in (and I have to configure Sat Tracker again), "Update" and "Save".

Now I return to the S20-24 preset, and the Map and Sat Tracker features have reappeared.

So in summary, there is something confusing me (or it is broken) regarding the state and interaction of Features and Presets.

srcejon, 2021-11-19:
Feature presets are currently separate from channel presets. In the top of the Features window, next to the add feature button, is a star for saving/loading Feature presets. Could add an enhancement request for this.

V96GLF, 2021-11-19:
Aha! I've never pressed that star ;-)

Deleted my Sat Tracker Channel preset; created a Feature preset, and resuming testing...

srcejon, 2021-11-19:
Not crashing for me with Ubuntu 21.10 / Qt 5.15.2.

Aside from the Sat Tracker and Map, did you have any other Features? What was in your Sat Tracker preset?

V96GLF, 2021-11-19:
Well, because I didn't understand the difference between Channel and Feature Presets, it probably doesn't contain anything at all!

I kept a copy...

[sat.zip](https://github.com/f4exb/sdrangel/files/7570825/sat.zip)
..

srcejon, 2021-11-19:
That's just what I needed, thanks.

It seems it can crash if the Map feature is in a window that is undocked. It will crash reproducibly, for me, just by undocking and then docking the Features window containing the Map. The crash only happens for Qt 5.15.2, not for Qt 5.14.2. It doesn't crash on Windows, but it doesn't work properly either once undocked.

It can also crash when undocking the ADS-B window (which contains a Map).

srcejon, 2021-11-19:
Looks a bit like this bug: https://bugreports.qt.io/browse/QTBUG-85260 which marked as fixed in Qt 5.15.3

closed 2022-02-24 by github-actions[bot]