f4exb/sdrangel #847: Enhancement: Add support for LoRa-Satellite Telemetry

df6dbf, 2021-04-14:
Since a few Weeks there are Nanosats with LoRa-Telemetry active.

https://tinygs.com

Suggestion for further Releases is to add a plugin for these Satellites.

73s Holger

f4exb, 2021-04-23:
- How is the data payload constructed?
  - What are the LoRa RF parameters to select? Bandwidth, Spread Factor, ...

None of these basic questions seem to be answered in the link provided or the Github repo link found inside or it is very obfuscated. The LoRa plugins (ChirpChat with LoRa option set that is...) already exist so this would be either adding some specifics to these plugins or make a feature plugin that deals just with the payload. In any case the issue is to get clear enough information on this payload.

df6dbf, 2021-04-24:
Asked for the specs in the tinygs Telegram Group and got the following answer:

Payload Decoder for Norby Satellite  is here:

https://www.youtube.com/watch?v=_Z69D0rPKBc&t=6s

ksy. File in the Comment and run with

https://ide.kaitai.io

LoRa RF Parameters for Norby (taken from Groundstationconfiguration)

{"mode":"LoRa","freq":436.703,"bw":250.0,"sf":10,"cr":5,"sw":18,"pwr":5,"cl":120,"pl":8,"gain":0,"crc":true,"fldro":1,"sat":"Norbi","NORAD":46494}

I not a Software Guy but i go with you. The Documentation is a little bit weird.

73s Holger

f4exb, 2021-06-15:
Looks like parsing the payload to interpret its contents thus I would say better implemented as a feature plugin.

f4exb, 2021-12-21:
.ksy file here: https://www.qsl.net/k/k4kdr//downloads/norbi/norbi.ksy

f4exb, 2022-02-10:
dropped