f4exb/sdrangel #848: Spurs in transmitted signal

sh123, 2021-04-15:
When transmitting single tone or CW in SSB modulator output signal has lots of spurs every ~40 kHz.

[image]

f4exb, 2021-04-15:
This is -60 dBc and you cannot expect better with this technology. The SSB modulator is based on audio sample rate which is usually 48 kHz. To minimize spurs you can use an integer multiple of this sample rate in the baseband (96, 192, 384, ... kS/s)

There is no action to be taken hence closing this issue.

closed 2021-04-15 by f4exb

sh123, 2021-05-04:
@f4exb , I think this problem somehow related to interpolation, when I'm using interpolation of 64 then this issue exists, when I'm using interpolation of 32 and less then this issue does not exist.