f4exb/sdrangel #849: sdrangel version  6.9.0-1 (Ubuntu package) crashing with floating point exception.

computer-wizard, 2021-04-16:
sdrangel version 6.9.0-1 (Ubuntu deb package) crashing with 'Floating point exception (core dumped)'
O/S: Ubuntu 20.04
This happens when the datv tx modulator is setup for a certain Symbol rate and Sample rate and when the start button is activated
after that.  SDR used is Pluto.  This doesn't happen for all symbol rates..  (Note Sample rate is always set to a multiple of the symbol rate)  For instance, it didn't happen with 2300 kS/s  DVB-S2

srcejon, 2021-04-16:
Which rate did it happen with?

computer-wizard, 2021-04-16:
Try  333 kS/s  with sample rate 3,330,000 S/s (10 x)
     1000 kS/s  with sample rate 3,000,000 S/s ( 3x)
     But there are  more

srcejon, 2021-04-17:
I can't reproduce this. I don't have a Pluto though.  It might possibly to be related to the TS as well (does it happen if you haven't selected a TS file?).

Could you perhaps send a screenshot showing all settings?

Also, you could try running gdb debugger to see where the exception is:

    gdb /opt/install/sdrangel/bin/sdrangel
    run

Then after it crashes type:

    info stack

And copy and paste the console text here.

computer-wizard, 2021-04-17:
I didn't select a TS file. What I did was just added a sink device set ->  Select sampling device (Pluto) -> Select Datv modulator, left every setting at default as the datv modulator came up with -> Changed the Symbols/s setting -> Set the proper sample rate ->
Set frequency -> click start.. At that stage stage it crashed right away. 

I already tried it with gdb before but there are no debugging symbols in the binary as supplied in the deb package  :-(
Anyhow here is, info stack

Thread 39 "QThread" received signal SIGFPE, Arithmetic exception.
[Switching to Thread 0x7fff897fa700 (LWP 62194)]
0x00007fffa3c25cb3 in DATVModSource::applySettings(DATVModSettings const&, bool) () from /usr/lib/sdrangel/plugins/libmoddatv.so
(gdb) info stack
#0  0x00007fffa3c25cb3 in DATVModSource::applySettings(DATVModSettings const&, bool) () at /usr/lib/sdrangel/plugins/libmoddatv.so
#1  0x00007fffa3c26d68 in DATVModBaseband::applySettings(DATVModSettings const&, bool) ()
    at /usr/lib/sdrangel/plugins/libmoddatv.so
#2  0x00007fffa3c2a055 in DATVModBaseband::handleMessage(Message const&) () at /usr/lib/sdrangel/plugins/libmoddatv.so
#3  0x00007fffa3c2a3ab in DATVModBaseband::handleInputMessages() () at /usr/lib/sdrangel/plugins/libmoddatv.so
#4  0x00007ffff6435d5a in QObject::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007ffff6e21a66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#6  0x00007ffff6e2b0f0 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#7  0x00007ffff640993a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007ffff640c5b8 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff6461f67 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007ffff4ba217d in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#11 0x00007ffff4ba2400 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#12 0x00007ffff4ba24a3 in g_main_context_iteration () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#13 0x00007ffff6461565 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff64084db in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00007ffff6240785 in QThread::exec() () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16 0x00007ffff62419d2 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#17 0x00007ffff53e9609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#18 0x00007ffff7ea7293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb)

srcejon, 2021-04-17:
Thanks - should be fixed in the next release.

To avoid it for now, make sure when changing the symbol rate, that the value is never 0. i.e. to change from 250000 to 1000000, add the 1 before deleting the 25.

computer-wizard, 2021-04-18:
Super,thanks!

f4exb, 2021-04-19:
Implemented in v6.10.0

closed 2021-04-19 by f4exb