f4exb/sdrangel #852: Datv modulator UDP port not  reachable

computer-wizard, 2021-04-19:
SDRangel 6.9.0-1 O/S: Ubuntu 20.04

Datv modulator udp port not reachable

Example:  127.0.0.1  port 5004  (and others at 127.0.0.1)
A ICMP packet is received with, udp port unreachable.

127.0.0.1 > 127.0.0.1: ICMP 127.0.0.1 udp port 5004 unreachable

Sdrangel  doesn't show up when the 'netstat' command is executed.

f4exb, 2021-04-19:
[image]

```
sudo netstat -anlpu                                                                                                                                                               
Active Internet connections (servers and established)                                                                                                                                                                                       
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name                                                                                                                                            
udp        0      0 0.0.0.0:9091            0.0.0.0:*                           806489/sdrangel                                                                                                                                             
...
udp        0      0 127.0.0.1:5004          0.0.0.0:*                           825484/./sdrangel                                                                                                                                           
...
udp6       0      0 :::6666                 :::*                                2510/qlipper          
```

computer-wizard, 2021-04-19:
Yes, I did here also but it is not listed
netstat -anlpu 
(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
udp        0      0 0.0.0.0:8554            0.0.0.0:*                           40427/obs           
udp        0      0 0.0.0.0:57733           0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:45649           0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:631             0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:53915           0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:2049            0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:51724           0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:40834           0.0.0.0:*                           -                   
udp        0      0 10.0.3.1:53             0.0.0.0:*                           -                   
udp        0      0 10.42.0.1:53            0.0.0.0:*                           -                   
udp        0      0 127.0.0.53:53           0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:67              0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:67              0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:67              0.0.0.0:*                           -                   
udp        0      0 192.168.2.10:68         192.168.2.1:67          ESTABLISHED -                   
udp        0      0 192.168.0.15:68         192.168.0.1:67          ESTABLISHED -                   
udp        0      0 0.0.0.0:69              0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:111             0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:37027           0.0.0.0:*                           -                   
udp6       0      0 fe80::53ac:d2c1:144:546 :::*                                -                   
udp6       0      0 :::58199                :::*                                -                   
udp6       0      0 :::50052                :::*                                -                   
udp6       0      0 :::54194                :::*                                -                   
udp6       0      0 :::5353                 :::*                                -                   
udp6       0      0 :::34186                :::*                                -                   
udp6       0      0 :::58831                :::*                                6628/skypeforlinux  
udp6       0      0 :::1716                 :::*                                6443/kdeconnectd    
udp6       0      0 :::2049                 :::*                                -                   
udp6       0      0 ::1:35514               ::1:48250               ESTABLISHED -                   
udp6       0      0 :::43825                :::*                                -                   
udp6       0      0 ::1:48250               ::1:35514               ESTABLISHED -                   
udp6       0      0 :::57142                :::*                                -                   
udp6       0      0 :::69                   :::*                                -                   
udp6       0      0 :::111                  :::*            

[image]

computer-wizard, 2021-04-19:
It also doesn't show in the just release version  6.10.0-1
(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
udp        0      0 0.0.0.0:57733           0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:45649           0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:631             0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:53915           0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:2049            0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:51724           0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:40834           0.0.0.0:*                           -                   
udp        0      0 10.0.3.1:53             0.0.0.0:*                           -                   
udp        0      0 10.42.0.1:53            0.0.0.0:*                           -                   
udp        0      0 127.0.0.53:53           0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:67              0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:67              0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:67              0.0.0.0:*                           -                   
udp        0      0 192.168.2.10:68         192.168.2.1:67          ESTABLISHED -                   
udp        0      0 192.168.0.15:68         192.168.0.1:67          ESTABLISHED -                   
udp        0      0 0.0.0.0:69              0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:111             0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:37027           0.0.0.0:*                           -                   
udp6       0      0 fe80::53ac:d2c1:144:546 :::*                                -                   
udp6       0      0 :::58199                :::*                                -                   
udp6       0      0 :::50052                :::*                                -                   
udp6       0      0 :::54194                :::*                                -                   
udp6       0      0 :::5353                 :::*                                -                   
udp6       0      0 :::34186                :::*                                -                   
udp6       0      0 :::58831                :::*                                6628/skypeforlinux  
udp6       0      0 :::1716                 :::*                                6443/kdeconnectd    
udp6       0      0 :::2049                 :::*                                -                   
udp6       0      0 ::1:35514               ::1:48250               ESTABLISHED -                   
udp6       0      0 :::43825                :::*                                -                   
udp6       0      0 ::1:48250               ::1:35514               ESTABLISHED -                   
udp6       0      0 :::57142                :::*                                -                   
udp6       0      0 :::69                   :::*                                -                   
udp6       0      0 :::111                  :::*                                -                   
[image]

f4exb, 2021-04-19:
Errmm.. you should select "UDP" there...
[image]

computer-wizard, 2021-04-19:
eh...ooooooooooooooops!

closed 2021-04-19 by computer-wizard