f4exb/sdrangel #853: DAB / DAB + demodulator

choupette-change, 2021-04-19:
Hello everyone, the DAB / DAB + demodulator does not appear in the Windows version ??? but ok version Ubuntu 20 10. Another demodulator to provide AIS VHF position of ships at sea. Good luck

srcejon, 2021-04-19:
Looks like neither demoddab.dll nor dab_lib.dll are installed.

It does when I build my own installer - so not sure what the problem might be. Perhaps not detecting zlib or faad when building?

f4exb, 2021-04-19:
Spotted this in the latest log of the MSVC build in Appveor:
```
[00:00:41] -- Could NOT find ZLIB (missing: ZLIB_LIBRARY ZLIB_INCLUDE_DIR) 
[00:00:42] -- Can't build DAB library as missing zlib or faad
```
This seems to confirm that zlib or faad is not detected.

srcejon, 2021-04-19:
Ok, I think I see the problem. find_package(zlib) added for Linux is overwriting the manual setting of ZLIB_FOUND for WIN32. Patch shortly.

IrishCoffee17, 2021-04-19:
Can confirm the new .exe installs the DAB demod plugin. Application crashes every time after selecting service from a DAB ensemble. Sometimes 1-2 seconds proper audio audible prior to crash. Right click onto speaker symbol in DAB demod shows no reaction. Windows 10 Home x64. In any case amazing work, guys.

IrishCoffee17, 2021-04-19:
Found reason for application crash. Had forgot to adjust sample rate to exactly 2.048MHz.

f4exb, 2021-04-19:
I think you need at least 2.048 MHz a higher value would do also.

IrishCoffee17, 2021-04-20:
Indeed it seems to work better with 2.048 MHz or multiples (4.096), but still not stable. Application always closes some seconds after selecting any audio program from any DAB or DAB+ ensemble.  Not much hints in the .log - btw is it normal, that .log is only written if application is run in elevated mode? Windows is sooo lovely... I am running my tests with PlutoSDR, SDRplay RSP2 and a cheap RTL-SDR stick.

IrishCoffee17, 2021-04-20:
I made further observations with the DAB demodulator which I better log as separate ticket, as they do not seem to be related to the initial installation process.

srcejon, 2021-04-20:
Whether you need elevation to write the log will depend upon the location of the log file (and possibly the permissions of the existing file). You will need elevation if the log is created in c:\program files\sdrangel - but not in c:\users\you (unless perhaps the log was originally created when run with elevation). Trying pointing the log to c:\users\you\sdrangel.log and make sure to delete all existing logs before running.

IrishCoffee17, 2021-04-20:
Thanks for clear explanation. That is exactly the case. Now I know how to solve.

choupette-change, 2021-04-20:
ok i see the DAB / DAB demodulator in the Windows.exe last version but after selecting a DAB radio it works fine but sdrangel crashes after about 20s of listening. I am on rtlsdr at SR 2048 MHz with a good antenna (this is ok with Welle.io for example) same crash problem on Ubuntu 20 10.

srcejon, 2021-04-20:
Are you able to run under gdb on linux to get a stack track when it crashes?

    sudo apt-get install gdb
    gdb /opt/install/sdrangel/bin/sdrangel
    run

After the crash

    info stack

Then copy and paste the output here?

srcejon, 2021-04-20:
Does it happen on every channel? For every program? For every sample rate? For DAB and DAB+?

choupette-change, 2021-04-20:
SDRAngel 6.10.1 installed with .deb Ubuntu 20 10
The scan of DAB+ ok (channel 12A)

radio Dab+ crash 20 s
2021-04-20 13:28:56.961 (D) DABDemodSink::applySettings:  force:  false
2021-04-20 13:28:56.961 (D) RTLSDRGui::handleInputMessages: message: RTLSDRInput::MsgConfigureRTLSDR
2021-04-20 13:28:56.962 (D) RTLSDRGui::updateFrequencyLimits: delta: 0 min: 24000 max: 1900000
[New Thread 0x7fff75ffb640 (LWP 4414)]
[Thread 0x7fff75ffb640 (LWP 4402) exited]
protection handler is eep_protection
we have now DAB+
[New Thread 0x7fff5d2fa640 (LWP 4415)]
2021-04-20 13:28:57.514 (D) DABDemodSink::applyDABAudioSampleRate: m_audioSampleRate: 48000 new m_dabAudioSampleRate: 48000
2021-04-20 13:28:57.514 (D) Interpolator::createPolyphaseLowPass: ntaps: 72
2021-04-20 13:28:57.514 (D) Ssample rate:  48000
2021-04-20 13:28:57.755 (W) QOpenGLTexture::setData() tried to set a null image
--Type <RET> for more, q to quit, c to continue without paging--

Thread 45 "QThread" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff5d2fa640 (LWP 4415)]
0x00007ffff7e3bc95 in __GI__IO_fwrite (buf=0x7fff6402e550, size=1, count=15655, fp=0x0)
    at iofwrite.c:35
35	iofwrite.c: Aucun fichier ou dossier de ce type.
(gdb) info stack
#0  0x00007ffff7e3bc95 in __GI__IO_fwrite (buf=0x7fff6402e550, size=1, count=15655, fp=0x0)
    at iofwrite.c:35
#1  0x00007fffb464a6ac in motObject::handleComplete() () at /usr/lib/sdrangel/libdab_lib.so
#2  0x00007fffb464aa33 in motObject::addBodySegment(unsigned char*, short, int, bool) ()
    at /usr/lib/sdrangel/libdab_lib.so
#3  0x00007fffb464d0d9 in padHandler::add_MSC_element(std::vector<unsigned char, std::allocator<unsigned char> >) () at /usr/lib/sdrangel/libdab_lib.so
#4  0x00007fffb464e576 in padHandler::handle_variablePAD(unsigned char*, short, unsigned char) ()
    at /usr/lib/sdrangel/libdab_lib.so
#5  0x00007fffb46403cd in mp4Processor::processSuperframe(unsigned char*, short) ()
    at /usr/lib/sdrangel/libdab_lib.so
#6  0x00007fffb46406d7 in mp4Processor::addtoFrame(unsigned char*) ()
    at /usr/lib/sdrangel/libdab_lib.so
#7  0x00007fffb4636efc in audioBackend::run() () at /usr/lib/sdrangel/libdab_lib.so
#8  0x00007ffff5fead84 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007ffff5393590 in start_thread (arg=0x7fff5d2fa640) at pthread_create.c:463
#10 0x00007ffff7ed5223 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb)

choupette-change, 2021-04-20:
crash without gdb :
2021-04-20 14:29:02.059 (D) DABDemodBaseband::handleMessage: MsgConfigureDABDemodBaseband
2021-04-20 14:29:02.059 (D) RTLSDRGui::handleInputMessages: message: RTLSDRInput::MsgConfigureRTLSDR
2021-04-20 14:29:02.059 (D) DABDemodSink::applySettings:  force:  false
2021-04-20 14:29:02.060 (D) RTLSDRGui::updateFrequencyLimits: delta: 0 min: 24000 max: 1900000
protection handler is eep_protection
we have now DAB+
2021-04-20 14:29:02.566 (D) DABDemodSink::applyDABAudioSampleRate: m_audioSampleRate: 48000 new m_dabAudioSampleRate: 48000
2021-04-20 14:29:02.566 (D) Interpolator::createPolyphaseLowPass: ntaps: 72
2021-04-20 14:29:02.567 (D) Ssample rate:  48000
2021-04-20 14:29:06.595 (W) QOpenGLTexture::setData() tried to set a null image
Erreur de segmentation (core dumped)

srcejon, 2021-04-20:
Thanks - that looks deep inside the DAB library.

df6dbf, 2021-04-20:
Crashed here also after a few seconds on Win10pro 64Bit  I5 6. Generation w 8GB Ram. Tried with RSPduo, RTLSDR (R820T) and PlutoSDR as Frontend.

73s Holger, DF6DBF

f4exb, 2021-04-21:
If issue is with the DAB library it must be for MSVC only. I can let it run for hours in Linux/Ubuntu.

srcejon, 2021-04-21:
Well, the stack trace from choupette-change is from Linux.

As we're all in different parts of the world, perhaps there is something different in the DAB broadcast that isn't supported? 

Would anyone be able to use the FileSink to record an .sdriq file so we can try to recreate it?

f4exb, 2021-04-21:
Good point. I haven't  tried many programs in the same feed either just my favorite ...

IrishCoffee17, 2021-04-21:
I recorded a 60 second .sdriq file of a DAB+ ensemble. Most of the time the app closes during file playback within these 60 seconds. Sometimes it takes a bit longer, so the best is to let the file loop by the app. Sooner or later it does crash. If that sounds interesting, how to share that 1 GB file?

srcejon, 2021-04-21:
Yep, great. I'd suggesting zipping the file then putting on something like Google Drive, Dropbox or OneDrive if you have an account.

IrishCoffee17, 2021-04-21:
[https://www.filemail.com/d/srhaqxaxeiqlxof](url) Hope it works

IrishCoffee17, 2021-04-21:
Alternative link [https://drive.google.com/file/d/1qrnqsV2RCVjtwAjxJjKiycex5oD2zMkT/view?usp=sharing](url)

srcejon, 2021-04-21:
Yep - got it thanks.

f4exb, 2021-04-21:
Yes got it and I've player the file ~20 times around without crash.

srcejon, 2021-04-21:
I've found one problem that can cause a crash, but will only effect windows build. Patch shortly.

srcejon, 2021-04-21:
Patch is on my dab-lib fork - which should fix the Windows crash.

However, looks like there is another problem. It doesn't result in a crash, but when playing Radio RADIESCHEN using a debug build, Visual Studio reports stack corruption in reedSolomon::decode_rs.

srcejon, 2021-04-21:
I've added a possible workaround for the stack corruption issue and reported the bug upstream.

IrishCoffee17, 2021-04-22:
@srcejon many thanks for your findings. Due to a lack of a local build environment I personally need to wait for the next available Windows .exe. But I will try to set something up for future opportunities. Found the excellently documentes Linux build instruction and might follow that unless you have a better recommendation, preferrably for Windows. But that is off-topic.

srcejon, 2021-04-22:
@IrishCoffee17 I've added Windows build instructions in issue #864

choupette-change, 2021-04-29:
I compiled by the sources under Ubuntu 20 10, lastest release v6.10.2
I am running my tests with SDRplay RSP1A
the same dongle, multiplex and radio station is ok with latest version Windows .exe 
Without gdb
021-04-29 13:17:53.820 (D) DABDemodSink::applySettings:  force:  false
2021-04-29 13:17:53.908 (D) SDRPlayV3Gui::updateHardware
2021-04-29 13:17:53.908 (D) SDRPlayV3Input::handleMessage: MsgConfigureSDRPlayV3
2021-04-29 13:17:53.908 (D) SDRPlayV3Input::applySettings
protection handler is eep_protection
we have now DAB+
Erreur de segmentation (core dumped)

with gdb
gdb /opt/install/sdrangel/bin/sdrangel
……….
protection handler is eep_protection
we have now DAB+
[New Thread 0x7fff77ffe640 (LWP 2791)]
2021-04-29 13:30:51.995 (W) QOpenGLTexture::setData() tried to set a null image
--Type <RET> for more, q to quit, c to continue without paging--

Thread 35 "QThread" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff77ffe640 (LWP 2791)]
0x00007ffff5ee6c95 in __GI__IO_fwrite (buf=0x7fff6802e1e0, size=1, 
    count=15655, fp=0x0) at iofwrite.c:35
35	iofwrite.c: Aucun fichier ou dossier de ce type.
(gdb)

srcejon, 2021-04-29:
@choupette-change are you able to create a .sdriq file so we can reproduce the problem?

choupette-change, 2021-04-29:
I created a 915.1 MB .sdriq but how do I send it to you on the github?

srcejon, 2021-04-29:
It's too big for github, so you'll need to use something like DropBox, GoogleDrive, OneDrive, FileMail etc. Thanks,

choupette-change, 2021-04-29:
https://drive.google.com/file/d/1fjwEoFuYEDeAtVVKUYeUpzsYiHLx6uJL/view?usp=sharing
i saved this file with the latest sdrangel version on windows. Tested under Ubuntu 20 10, it crashes around the 25th second but it's random, tested with the "BONHEUR" station. Stations of the multiplex work correctly but sometimes the logo of the radio station does not always appear ..... I also use Welle.io and everything is ok with Ubuntu 20 10

srcejon, 2021-04-29:
Thanks, I can reproduce the failure here.

f4exb, 2021-04-29:
Here to on Ubuntu 20.10 with "BONHEUR" station.

f4exb, 2021-05-01:
I am now running the IQ file fine with "BONHEUR" program. Thanks!

f4exb, 2021-05-01:
Latest fix deployed in v6.10.3

closed 2021-05-01 by f4exb

danwdart, 2022-07-03:
Hmm... v6.18.1 is missing DAB/DAB+ on NixOS, but I'm not sure what to add here: https://github.com/NixOS/nixpkgs/blob/nixos-22.05/pkgs/applications/radio/sdrangel/default.nix - anybody know? Thanks

srcejon, 2022-07-03:
Probably the dab library not built: https://github.com/srcejon/dab-cmdline.git

Not familar with that build system though.