f4exb/sdrangel #855: DAB/DAB+ demodulator

IrishCoffee17, 2021-04-20:
Windows 10 Home x64, sdrangel v6.10.1, tested on 2 different PCs, input device is currently an SDRplay RSP2 with API 3.07.2
App is closing every time if SNR drops below 16 whilst listening to any DAB/DAB+ program
In general I love this app, but my own programming skills are unfortunately not that good.

2021-04-20 08:55:36.740 (D) SDRPlayV3Input::handleMessage: MsgConfigureSDRPlayV3
2021-04-20 08:55:36.740 (D) SDRPlayV3Input::applySettings
2021-04-20 08:55:36.740 (D) ChannelWebAPIUtils::setCenterFrequency: set device frequency 220352000.000000 OK
2021-04-20 08:55:36.740 (D) DABDemod::handleMessage: MsgConfigureDABDemod
2021-04-20 08:55:36.741 (D) DABDemod::applySettings:  m_streamIndex:  0  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0  m_reverseAPIChannelIndex:  0  force:  false
2021-04-20 08:55:36.741 (D) DABDemodBaseband::handleMessage: MsgConfigureDABDemodBaseband
2021-04-20 08:55:36.741 (D) DABDemodSink::applySettings:  force:  false
2021-04-20 08:55:36.741 (D) SDRPlayV3Gui::handleInputMessages: message: SDRPlayV3Input::MsgConfigureSDRPlayV3
2021-04-20 08:55:36.838 (D) SDRPlayV3Gui::updateHardware
2021-04-20 08:55:36.839 (D) SDRPlayV3Input::handleMessage: MsgConfigureSDRPlayV3
2021-04-20 08:55:36.839 (D) SDRPlayV3Input::applySettings
2021-04-20 08:55:37.281 (D) DABDemodSink::applyDABAudioSampleRate: m_audioSampleRate: 48000 new m_dabAudioSampleRate: 48000
2021-04-20 08:55:37.281 (D) Interpolator::createPolyphaseLowPass: ntaps: 72
2021-04-20 08:55:37.282 (D) Ssample rate:  48000
2021-04-20 08:55:43.932 (D) SDRPlayV3Gui::updateHardware
2021-04-20 08:55:43.932 (D) SDRPlayV3Input::handleMessage: MsgConfigureSDRPlayV3
2021-04-20 08:55:43.932 (D) SDRPlayV3Input::applySettings
2021-04-20 08:55:48.886 (D) SDRPlayV3Gui::updateHardware
2021-04-20 08:55:48.886 (D) SDRPlayV3Input::handleMessage: MsgConfigureSDRPlayV3
2021-04-20 08:55:48.886 (D) SDRPlayV3Input::applySettings
2021-04-20 08:56:00.574 (D) SDRPlayV3Gui::updateHardware
2021-04-20 08:56:00.574 (D) SDRPlayV3Input::handleMessage: MsgConfigureSDRPlayV3
2021-04-20 08:56:00.574 (D) SDRPlayV3Input::applySettings
2021-04-20 08:56:02.610 (C) SampleSinkFifo::write: 481 messages dropped
2021-04-20 08:56:02.610 (C) SampleSinkFifo::write: overflow - dropping 384 samples
2021-04-20 08:56:02.610 (C) SampleSinkFifo::write: overflow - dropping 1344 samples

srcejon, 2021-04-20:
Hmm, it runs OK for me on Windows when SNR is lower than 16. (It can go down to 4 or 5, but the audio will sound poor)

Not sure what to suggest. Can you send a screenshot so we can see all your settings?

IrishCoffee17, 2021-04-20:
Interesting. I admit that I have a poor indoor antenna connected, but that should not matter for DAB. Here a screenshot:

[image]

srcejon, 2021-04-20:
That all looks fine. What CPU do you have?

IrishCoffee17, 2021-04-20:
[image]

IrishCoffee17, 2021-04-23:
Symptom is gone with the latest fixes. Currently no instabilities whatsoever on the windows build. Closing issue.

closed 2021-04-23 by IrishCoffee17