f4exb/sdrangel #858: AIS Demodulator

choupette-change, 2021-04-20:
Evolution of a new demodulator: Have you planned to make an ais demodulator (vhf position of ships), like the adsb demodulator, soon, with the card already present in sdrangel ? I am already using "aisdecoder" but everything integrated into sdrangel that would be very good...

choupette-change, 2021-05-09:
I test the AIS demodulator + feature on Ubuntu 20 10 (rtlsdr) it's ok for me (I live near the west coast of France) . Where did you find the icons? Good work . Can you predict the units (km, s, ..)?

srcejon, 2021-05-09:
If you can find/make some better and more diverse icons, I will update them. I couldn't find many that were top down, which is needed to draw them according to their heading.

f4exb, 2021-05-11:
Works great!
[image]

srcejon, 2021-05-11:
The names don't look right though. The @ character is a 0.  I strip them from the start and end, but hadn't seen any in the middle, which is a little suspect (although the callsign etc seem ok). If you can make an sdriq file, I can take a look. Thanks.

srcejon, 2021-05-11:
Actually, they all look like small boats rather than ships, which means the problem is with Class B messages, and I think I see the problem.

srcejon, 2021-05-11:
Thanks - looks like there's another problem with Static Data Reports. Not such a variety of message type around here.

choupette-change, 2021-05-11:
ship names seem correct to me
[image]
[image]

Valstee, 2021-05-12:
So Nice, Thanks Srcejon !

f4exb, 2021-06-12:
Implemented

closed 2021-06-12 by f4exb