f4exb/sdrangel #860: Exception when closing channel analyzer

srcejon, 2021-04-20:
When debugging using Visual Studio on a debug build, closing the Channel Analyzer throws an exception (I think because the debugger invalidates pointers to deleted objects).

It seems to come from:

    ChannelAnalyzerGUI::~ChannelAnalyzerGUI()
        {
            m_channelAnalyzer->setSampleSink(nullptr);

which has the code:

    void setSampleSink(BasebandSampleSink *sink) { m_basebandSink->setSampleSink(sink); }

However, in the log file, it appears that m_basebandSink has already been deleted at this point:

2021-04-20 16:42:04.117 (D) ChannelGUI::closeEvent
2021-04-20 16:42:04.119 (D) ChannelAnalyzer::~ChannelAnalyzer
2021-04-20 16:42:04.119 (D) DSPDeviceSourceEngine::removeSink:  ChannelAnalyzer
2021-04-20 16:42:04.120 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPRemoveBasebandSampleSink
2021-04-20 16:42:04.120 (D) ChannelAnalyzerBaseband::~ChannelAnalyzerBaseband   <----------
2021-04-20 16:42:04.120 (D) FFTFactory::releaseEngine: engineSequence: 0 FFT fwd size: 8192
2021-04-20 16:42:04.121 (D) FFTFactory::releaseEngine: engineSequence: 1 FFT fwd size: 8192
2021-04-20 16:42:04.121 (D) FFTFactory::releaseEngine: engineSequence: 0 FFT inv size: 8192
2021-04-20 16:42:04.122 (D) ChannelAnalyzer::~ChannelAnalyzer: done
2021-04-20 16:42:04.122 (D) FFTFactory::releaseEngine: engineSequence: 0 FFT fwd size: 1024

f4exb, 2021-04-20:
This was put in place to solve issue #806 maybe the setting to `nullptr` is just misplaced in the destructor of the GUI.

When the close event occurs by closing the GUI with the "X" the `ChannelGUI` interface re-emits a `closing` event that is connected in the `DeviceUISet` to its `handleChannelGUIClosing` slot. This in turn calls the  `destroy` API of the `ChannelAPI` interface that results in the destructor of the channel to be effectively called. then there is a race condition with the destructor of the GUI. Therefore the `nullptr` setting should rather be on the channel destructor path rather than in its GUI.

f4exb, 2021-04-20:
If the `closeEvent` in `ChannelGUI` is handled like this:
```c++
void ChannelGUI::closeEvent(QCloseEvent *event)
{
    qDebug("ChannelGUI::closeEvent");
    destroy();
    emit closing();
    event->ignore();
}
```
Then perhaps we can make sure the GUI is always deleted first.

f4exb, 2021-04-21:
```c++
    destroy();
    emit closing();
```
errm... this is what one would call "cutting off the branch you're sitting on."... doesn't work...

f4exb, 2021-04-21:
In fact the order in which the destructors of the "demod" and the GUI are called depend on the way the destruction of the channel is triggered and is opposite:

Via GUI (close event) - demod first then GUI:
```
2021-04-21 11:40:39.136 (D) ChannelGUI::closeEvent
2021-04-21 11:40:39.137 (D) ChannelAnalyzer::~ChannelAnalyzer
2021-04-21 11:40:39.137 (D) DSPDeviceSourceEngine::removeSink:  ChannelAnalyzer
2021-04-21 11:40:39.137 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPRemoveBasebandSampleSink
2021-04-21 11:40:39.137 (D) ChannelAnalyzer::stop
2021-04-21 11:40:39.137 (D) ChannelAnalyzerBaseband::~ChannelAnalyzerBaseband
2021-04-21 11:40:39.137 (D) FFTFactory::releaseEngine: engineSequence: 0 FFT fwd size: 8192
2021-04-21 11:40:39.137 (D) FFTFactory::releaseEngine: engineSequence: 1 FFT fwd size: 8192
2021-04-21 11:40:39.137 (D) FFTFactory::releaseEngine: engineSequence: 0 FFT inv size: 8192
2021-04-21 11:40:39.137 (D) ChannelAnalyzer::~ChannelAnalyzer: done
2021-04-21 11:40:39.137 (D) FFTFactory::releaseEngine: engineSequence: 0 FFT fwd size: 1024
2021-04-21 11:40:39.137 (D) ChannelAnalyzerGUI::~ChannelAnalyzerGUI
2021-04-21 11:40:39.137 (D) GLScope::disconnectTimer
2021-04-21 11:40:39.137 (D) ChannelAnalyzerGUI::~ChannelAnalyzerGUI: done
2021-04-21 11:40:39.142 (W) QOpenGLTexturePrivate::destroy() called without a current context.
Texture has not been destroyed
2021-04-21 11:40:39.142 (W) QOpenGLTexturePrivate::destroy() called without a current context.
Texture has not been destroyed
2021-04-21 11:40:39.142 (W) QOpenGLTexturePrivate::destroy() called without a current context.
Texture has not been destroyed
2021-04-21 11:40:39.142 (W) QOpenGLTexturePrivate::destroy() called without a current context.
Texture has not been destroyed
```

Via preset load - GUI first then demod:
```
2021-04-21 11:45:00.136 (D) MainWindow::on_presetLoad_clicked
2021-04-21 11:45:00.136 (D) MainWindow::loadPresetSettings: preset [AIR | LFMD info]
...
2021-04-21 11:45:00.190 (D) ChannelAnalyzerGUI::~ChannelAnalyzerGUI
2021-04-21 11:45:00.190 (D) DSPDeviceSourceEngine::handleInputMessages: message: DSPSignalNotification
2021-04-21 11:45:00.190 (D) GLScope::disconnectTimer
2021-04-21 11:45:00.190 (D) DSPDeviceSourceEngine::handleInputMessages: DSPSignalNotification:  m_sampleRate:  1000000  m_centerFrequency:  130480000
2021-04-21 11:45:00.190 (D) ChannelAnalyzerGUI::~ChannelAnalyzerGUI: done
...
2021-04-21 11:45:00.195 (D) ChannelAnalyzer::~ChannelAnalyzer
2021-04-21 11:45:00.195 (D) DSPDeviceSourceEngine::removeSink:  ChannelAnalyzer
2021-04-21 11:45:00.195 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPRemoveBasebandSampleSink
2021-04-21 11:45:00.195 (D) ChannelAnalyzer::stop
2021-04-21 11:45:00.195 (D) ChannelAnalyzerBaseband::~ChannelAnalyzerBaseband
2021-04-21 11:45:00.195 (D) FFTFactory::releaseEngine: engineSequence: 0 FFT fwd size: 8192
2021-04-21 11:45:00.196 (D) FFTFactory::releaseEngine: engineSequence: 1 FFT fwd size: 8192
2021-04-21 11:45:00.196 (D) FFTFactory::releaseEngine: engineSequence: 0 FFT inv size: 8192
2021-04-21 11:45:00.196 (D) ChannelAnalyzer::~ChannelAnalyzer: done
...
```
When fixing #806 only the preset case was considered.

f4exb, 2021-04-21:
In fact this order is reproduced in other cases in `DeviceUISet` that should be fixed the same way.

f4exb, 2021-04-21:
Same issue with `FeatureUISet`.

commit de23efe6 (2021-04-21): Delete channel or feature and their GUI in the proper order in DeviceUISet and FeatureUISet. Remove original fix for #806. Fixes #860
closed 2021-04-21 by f4exb