f4exb/sdrangel #861: Maintainin modAFInput state through PTT

N5BRG, 2021-04-21:
I am trying to complete a Hamlib interface for sdrangel and have a couple of issues to resolve.
I am finding that when I switch between R0->T1  or T1->R0 the modAFInput is changing to an inactive state. The REST process I am using is:
DELETE the run state on the device to be inactive
POST the run state on the device to become active
PATCH the newly active device to become in FOCUS.

When I use the Simple PTT feature the modAFInput maintains its state.

I have added additional code to force the modAFInput to the correct state when going into transmit and this works but the additional REST interface exchange is taking too much time.  When working with WSJTX that application times out and stops working while wainting on Hamlib to complete PTT.

The mic input must be active for WSJTX to modulate the transmission so this state is always required when using WSJTX.

Can this be improved in some way?

Is it possible to use the Simple PTT Feature via the REST interface? I do not see how to get this working through REST. Not sure modAFInput will remain sticky doing this but it may if you are making PTT changes in the code. I would like to make PTT faster also and this is a way to do it.

I am planning to release Hamlib code to support Pluto, HackRF, and RTL-SDR (receive only) first but would like to include more devices in time. I can post Hamlib code now but does not support WSJTX properly and that is main interest for automatic operation.

Any help would be appreciated.  I have one other issue that will be posted separately.

Bob N5BRG

N5BRG, 2021-04-21:
Using sdrangel version 6.10.1-3    Fresh build today
Working on Intel I5 based Ubuntu 20.10 system

f4exb, 2021-04-21:
> Is it possible to use the Simple PTT Feature via the REST interface? 

Yes this is provided by `/sdrangel/featureset` entry points. See http://127.0.0,1:8091/doc/swagger-ui/index.html#/FeatureSet with an SDRangel instance running with default values. Note that for now there are no multiple feature sets so the `featureSetIndex` is always 0.

However the "simple" PTT feature may be too simple for your purpose. See my other reply for #862

f4exb, 2021-04-21:
Regarding the original issue. DELETING and POSTING on `/sdrangel/deviceset/{deviceSetIndex}/device/run` endpoint should not affect the `modAFInput` in principle. It is another story if you tweak spmething with the settings or delete and re-create the Tx device set completely.

N5BRG, 2021-04-22:
OK, I will do more testing here. I have noticed the icon does not always remain active so I can enable   the MIC during transmit. I can right click on it and select default which will slect the MIC and start modulation. 

The tip on the feature set was helpful I had been looking in swagger but never looked with PTT started. What I see is

{
  "featurecount": 0,
  "features": [
    {
      "index": 0,
      "id": "string",
      "uid": 0,
      "title": "string"
    }
  ]
}

Don't see a way to do a PUSH to TALK on PTT with a PATCH of the feature? This would allow me to send one REST command in place of three if the feature would allow PTT.  

This may get people with a fast computer working but not support the RPi crowd if WSJTX is working on a true clock. They say PTT has to be done in 1 second but I have been running a timer and it looks like things keep working up to about 3 seconds. The REST interface time requirement seems to vary a lot, most of the time very fast but also dead slow sometimes.

I will continue to test. I just hate to release the code if I see errors on occasion. 

Again thanks for your feedback and efforts.

Bob N5BRG

N5BRG, 2021-04-22:
Post the wrong past:
{
  "featurecount": 1,
  "features": [
    {
      "id": "SimplePTT",
      "index": 0,
      "title": "Simple PTT",
      "uid": 1619024091100206
    }
  ]
}

f4exb, 2021-04-22:
I have checked and the `ModAF` input setting is not changed when the Simple PTT feature switches Rx/Tx back and forth.

N5BRG, 2021-04-22:
I am doing some of my work remotely with SSH and all the windows do not refresh properly. I need to get in from of the computer running the code and test there and even monitor the transmission in case my local system is fooling me. I hear the tones but the monitor scope does not have a trace and the MIC is not indicating active. Maybe it is working properly and the display on my computer is not.

N5BRG, 2021-04-28:
As you indicated as long as I am moving just between RX an TX with PTT the modAFInput is stable. The problem is WSJTX does other things like change TX frequency between RX and TX and when the other steps are inserted the modAFInput is changed. I am adding code to write the correct value to modAFInput but then I exceed the time limit of 1 second that WSJTX imposes.

I have not posted code to the world yet because of several issues i was trying to solve. I know you want to look at code and can not help me until i post. Probably will do this tomorrow my time.

Bob

f4exb, 2021-04-28:
If you do anything with the settings you have to use PATCH to change only the settings present in the JSON payload. If you use PUT it will use the default values for parameters not present in the JSON.

N5BRG, 2021-04-30:
When working with CURL I have had some issues limiting the settings to be changed.
This caused me to be worried about staying current with the user and pulling too much data to edit only the item I wanted to change.  I need to try using Python which seems to have a better JSON interface.
The modAFInput setting does seem to change though if a user does a frequency change between turning PTT on and off.

I understand all my problems and will close this issue.

Bob N5BRG

f4exb, 2021-05-01:
If you can use Qt there is the QJsonObject: https://doc.qt.io/qt-5/qjsonobject.html that makes it easier to manipulate JSON in C++. You can find it used in many places in SDRangel code. However Python is probably easier to deal with.