f4exb/sdrangel #864: Add Windows and Mac build instructions on Wiki

srcejon, 2021-04-22:
I think it would be useful to add Windows and Mac build instructions on a Wiki page.

Windows:

Install Visual Studio 2019 (Community edition is free - https://visualstudio.microsoft.com/vs/community/)
Install Qt 5.15.x (https://www.qt.io/download-open-source)
Install Boost (https://www.boost.org/users/download/)
Install ninja (https://ninja-build.org/)  
Install git (https://git-scm.com/download/win)

Make sure git and ninja are in your PATH environment variable.

Run Visual Studio shell: Start > Visual Studio 2019 > x64 Native Tools Command Prompt for VS 2019

    cd %userprofile% (or wherever you want to build that has 10GB free space)
    git clone https://github.com/f4exb/sdrangel.git
    cd sdrangel
    git submodule update --init --recursive
    mkdir build
    cd build
    cmake -Wno-dev -G Ninja -DCMAKE_BUILD_TYPE=Release -DENABLE_BLADERF=OFF -DENABLE_IIO=OFF -DENABLE_MIRISDR=OFF -DRX_SAMPLE_24BIT=ON -DCMAKE_PREFIX_PATH=C:\\Qt\\5.15.2\\msvc2019_64;C:\\boost_1_74_0 ..
    ninja
    ninja package

This will build the installer .exe

After an initial build of 'ninja package', rather than repeatedly building the installer, you can run SDRangel from the build directory, by just running 'ninja' to rebuild then '.\bin\sdrangel' to run.

Passing  -DCMAKE_BUILD_TYPE=Debug or  -DCMAKE_BUILD_TYPE=RelWithDebInfo to the cmake command will create a version that is easier to debug.

To debug using Visual Studio, run:

    devenv /DebugExe .\bin\SDRangel.exe 

Mac:

Install XCode from AppStore
Install MacPorts from: https://www.macports.org/install.php

In a terminal, run:

    sudo port selfupdate
    port search sdrangel
    sudo port install sdrangel
    sdrangel

More elaborate instructions probably needed for building directly from https://github.com/f4exb/sdrangel.git

IrishCoffee17, 2021-04-22:
Very useful indeed! 
VS 2019 developer prompt does not find cmake command. Any particular workloads to select during VS installation? Thanks.

srcejon, 2021-04-22:
Sorry, you need to install cmake as well: https://cmake.org/ - I'll update the text above.

IrishCoffee17, 2021-04-22:
Next challenge is a compiler error:

CMake Error at C:/Qt/5.15.2/mingw81_64/lib/cmake/Qt5/Qt5Config.cmake:28 (find_package):
  Could not find a package configuration file provided by "Qt5Charts" with
  any of the following names:

    Qt5ChartsConfig.cmake
    qt5charts-config.cmake

To solve, add the "QT Charts" to the Qt5.15.2 install option (checkbox in lower end of screenshot) and then rerun cmake.

[image]

srcejon, 2021-04-22:
Thanks - you should probably have MSVC 2019 64-bit checked there too (As that's what Visual Studio uses)

IrishCoffee17, 2021-04-22:
Good point, actually I had never consciously checked MinGW. But I will now do for MSVC 2019 64-bit.

f4exb, 2021-04-22:
Done adding some headers and cosmetic changes:
  - Windows: https://github.com/f4exb/sdrangel/wiki/Compile-in-Windows
  - Mac: https://github.com/f4exb/sdrangel/wiki/Build-for-Mac-using-Macports

Also referenced in the main right menu at the top: https://github.com/f4exb/sdrangel/wiki

IrishCoffee17, 2021-04-22:
Next challenge (Windows build): i come as far as the ninja command. Aborts as follows:

[46/2239] cmd.exe /C "cd /D C:\Users\xxxxx\sdrangel\build\external\soapysdr\src\soapysdr && "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -P C:/Users/xxxxx/sdrangel/build/external/soapysdr/tmp/soapysdr-gitupdate.cmake"
[47/2239] cmd.exe /C "cd /D C:\Users\xxxxx\sdrangel\build\external && "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -E echo_append && "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -E touch C:/Users/xxxxx/sdrangel/build/external/dsdcc/src/dsdcc-stamp/dsdcc-patch"
[48/2239] cmd.exe /C "cd /D C:\Users\xxxxx\sdrangel\build\external && "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -E echo_append && "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -E touch C:/Users/xxxxx/sdrangel/build/external/soapysdr/src/soapysdr-stamp/soapysdr-patch"
[49/2239] cmd.exe /C "cd /D C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr-build && "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" --build C:/Users/xxxxx/sdrangel/build/external/rtlsdr/src/rtlsdr-build --config Release -- rtlsdr_shared && "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -E touch C:/Users/xxxxx/sdrangel/build/external/rtlsdr/src/rtlsdr-stamp/rtlsdr-build"
**FAILED: external/rtlsdr/src/rtlsdr-stamp/rtlsdr-build lib/rtlsdr.lib** 
cmd.exe /C "cd /D C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr-build && "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" --build C:/Users/xxxxx/sdrangel/build/external/rtlsdr/src/rtlsdr-build --config Release -- rtlsdr_shared && "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -E touch C:/Users/xxxxx/sdrangel/build/external/rtlsdr/src/rtlsdr-stamp/rtlsdr-build"
[1/2] Building RC object src\CMakeFiles\rtlsdr_shared.dir\rtlsdr.rc.res
FAILED: src/CMakeFiles/rtlsdr_shared.dir/rtlsdr.rc.res 
"C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/bin/cmcldeps.exe" RC C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr-build\src\rtlsdr.rc src\CMakeFiles\rtlsdr_shared.dir\rtlsdr.rc.res.d src\CMakeFiles\rtlsdr_shared.dir\rtlsdr.rc.res "Hinweis: Einlesen der Datei: " "C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29910/bin/Hostx86/x86/cl.exe" C:\PROGRA~2\WI3CF2~1\10\bin\100190~1.0\x86\rc.exe -D_TIMESPEC_DEFINED -Drtlsdr_EXPORTS -I C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr\include -I C:\Users\xxxxx\sdrangel\external\windows\libusb\include\libusb-1.0 -I C:\Users\xxxxx\sdrangel\build\external\pthreads4w\src\pthreads4w -DWIN32 /fo src\CMakeFiles\rtlsdr_shared.dir\rtlsdr.rc.res C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr-build\src\rtlsdr.rc
cl : Befehlszeile warning D9035 : Die Option "o" ist veraltet und wird in einer der nächsten Versionen entfernt.

C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr-build\src\rtlsdr.rc(2): fatal error C1083: Datei (Include) kann nicht ge÷ffnet werden: "afxres.h": No such file or directory

**ninja: build stopped: subcommand failed.**

srcejon, 2021-04-22:
Not, sure, as I haven't seen that problem. One thing that looks odd is the reference to Hostx86/x86/cl.exe? Are you definitely in a x64 shell rather than x86?

When you start the shell, it should output:

**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.9.1
** Copyright (c) 2021 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

Check for x64 at the end.

Note, if you were accidentally in an x86 shell, best to completely delete your build directory and start again.

IrishCoffee17, 2021-04-22:
Oops you're right - 32bit...
Back to square_1

IrishCoffee17, 2021-04-22:
Hmm still not much better.
What I noted at this point is that "external/rtlsdr/src/rtlsdr-stamp" has no further subdirectories such as "rtlsdr-build"

[116/2288] Performing build step for 'rtlsdr'
FAILED: external/rtlsdr/src/rtlsdr-stamp/rtlsdr-build lib/rtlsdr.lib
cmd.exe /C "cd /D C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr-build && "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" --build C:/Users/xxxxx/sdrangel/build/external/rtlsdr/src/rtlsdr-build --config Release -- rtlsdr_shared && "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -E touch C:/Users/xxxxx/sdrangel/build/external/rtlsdr/src/rtlsdr-stamp/rtlsdr-build"
[1/8] Building C object src\CMakeFiles\rtlsdr_shared.dir\tuner_fc0012.c.obj
[2/8] Building C object src\CMakeFiles\rtlsdr_shared.dir\tuner_fc2580.c.obj
[3/8] Building C object src\CMakeFiles\rtlsdr_shared.dir\tuner_e4k.c.obj
C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr\src\tuner_e4k.c(448): warning C4244: "return": Konvertierung von "uint64_t" in "uint32_t", möglicher Datenverlust
C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr\src\tuner_e4k.c(518): warning C4244: "=": Konvertierung von "uint64_t" in "uint32_t", möglicher Datenverlust
C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr\src\tuner_e4k.c(521): warning C4244: "Funktion": Konvertierung von "uint64_t" in "uint8_t", möglicher Datenverlust
C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr\src\tuner_e4k.c(530): warning C4244: "=": Konvertierung von "uint64_t" in "uint8_t", möglicher Datenverlust
[4/8] Building C object src\CMakeFiles\rtlsdr_shared.dir\tuner_fc0013.c.obj
[5/8] Building C object src\CMakeFiles\rtlsdr_shared.dir\tuner_r82xx.c.obj
C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr\src\tuner_r82xx.c(488): warning C4244: "=": Konvertierung von "uint64_t" in "uint8_t", möglicher Datenverlust
C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr\src\tuner_r82xx.c(489): warning C4244: "=": Konvertierung von "uint64_t" in "uint32_t", möglicher Datenverlust
[6/8] Building C object src\CMakeFiles\rtlsdr_shared.dir\librtlsdr.c.obj
C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr\src\librtlsdr.c(704): warning C4244: "=": Konvertierung von "double" in "int32_t", möglicher Datenverlust
C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr\src\librtlsdr.c(720): warning C4244: "Initialisierung": Konvertierung von "double" in "int16_t", möglicher Datenverlust
C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr\src\librtlsdr.c(1110): warning C4244: "=": Konvertierung von "double" in "uint32_t", möglicher Datenverlust
[7/8] Building RC object src\CMakeFiles\rtlsdr_shared.dir\rtlsdr.rc.res
FAILED: src/CMakeFiles/rtlsdr_shared.dir/rtlsdr.rc.res
"C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/bin/cmcldeps.exe" RC C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr-build\src\rtlsdr.rc src\CMakeFiles\rtlsdr_shared.dir\rtlsdr.rc.res.d src\CMakeFiles\rtlsdr_shared.dir\rtlsdr.rc.res "Hinweis: Einlesen der Datei: " "C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29910/bin/Hostx64/x64/cl.exe" C:\PROGRA~2\WI3CF2~1\10\bin\100190~1.0\x64\rc.exe -DHAVE_LIBUSB_ERROR_NAME=1 -DHAVE_LIBUSB_HANDLE_EVENTS_TIMEOUT_COMPLETED=1 -D_TIMESPEC_DEFINED -Drtlsdr_EXPORTS -I C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr\include -I C:\Users\xxxxx\sdrangel\external\windows\libusb\include\libusb-1.0 -I C:\Users\xxxxx\sdrangel\build\external\pthreads4w\src\pthreads4w -DWIN32 /fo src\CMakeFiles\rtlsdr_shared.dir\rtlsdr.rc.res C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr-build\src\rtlsdr.rc
cl : Befehlszeile warning D9035 : Die Option "o" ist veraltet und wird in einer der n├ñchsten Versionen entfernt.
C:\Users\xxxxx\sdrangel\build\external\rtlsdr\src\rtlsdr-build\src\rtlsdr.rc(2): fatal error C1083: Datei (Include) kann nicht ge├Àffnet werden: "afxres.h": No such file or directory
ninja: build stopped: subcommand failed.
[121/2288] Building CXX object swagger\CMakeFiles\swagger.dir\sdrangel\code\qt5\client\SWGFeatureSetApi.cpp.obj
..\swagger\sdrangel\code\qt5\client\SWGFeatureSetApi.cpp(48): warning C4996: 'QByteArray::append': Use QString's toUtf8(), toLatin1() or toLocal8Bit()
..\swagger\sdrangel\code\qt5\client\SWGFeatureSetApi.cpp(107): warning C4996: 'QByteArray::append': Use QString's toUtf8(), toLatin1() or toLocal8Bit()
..\swagger\sdrangel\code\qt5\client\SWGFeatureSetApi.cpp(220): warning C4996: 'QByteArray::append': Use QString's toUtf8(), toLatin1() or toLocal8Bit()
..\swagger\sdrangel\code\qt5\client\SWGFeatureSetApi.cpp(559): warning C4996: 'QByteArray::append': Use QString's toUtf8(), toLatin1() or toLocal8Bit()
ninja: build stopped: subcommand failed.

IrishCoffee17, 2021-04-22:
In my case, the "x64 Native Tools Command Prompt for VS 2019" is only available, if the workload "Desktop development with C++" option was installed with MSVC 2019.

f4exb, 2021-04-22:
I've sent an invite to `srcejon` that should allow Wiki updates

IrishCoffee17, 2021-04-23:
Progress update:
- After disabling rtlsdr in CMakeLists.txt to bypass above mentioned build issue, build process finished nicely.
- please add nsis to installation prerequisite, add to path
- SDRangel installs nicely from .exe package 
- DAB demod instability currently unchanged. Should the patch be included? Working on useful debug outputs.

srcejon, 2021-04-23:
You don't have to disable drivers in CMakeLists.txt, you can do it from the command line with -DENABLE_RTLSDR=OFF. I'll add NSIS, thanks.

Yes, the patch should be in the DAB library sub-project. You can check by looking in 

sdrangel\build\external\dab\src\dab\library\src\backend\reed-solomon.cpp - Line 366 should be:

	memcpy (workRegister, Lambda, (nroots + 1) * sizeof (uint8_t));

Not

	memcpy (&workRegister, Lambda, (nroots + 1) * sizeof (uint8_t));

To debug, I suggest you create a new build directory, sdrangel/builddebug then run cmake there changing:

    -DCMAKE_BUILD_TYPE=Release

To:

    -DCMAKE_BUILD_TYPE=RelWithDebInfo

(-DCMAKE_BUILD_TYPE=Debug will probably be too slow)

Then run the debugger with:

    devenv /DebugExe .\bin\SDRangel.exe 

When it crashes, you should be able to see in which function it happens and send a screenshot with a stack trace.

IrishCoffee17, 2021-04-23:
I followed exactly your steps.
Patch is present. We have a result:

[image]

srcejon, 2021-04-23:
Looks like you are getting an image broadcast! And the code is trying to write it to a linux style path, without checking the file was opened successfully, which it wont on Windows. Should be easy to change.

IrishCoffee17, 2021-04-23:
If I understand correctly, this happens with the DAB MOT service (slide shows), which would explain, why DAB demod runs fine for about 20 seconds, which is a typical transmission time for a slide until it is fully fetched. Does this make sense?

IrishCoffee17, 2021-04-23:
Yes, MOT images are very popular here.

srcejon, 2021-04-23:
Didn't realise it was in use. I guess we need a way to display the pic in SDRangel too :) I'll have to see if I can find a UK station that uses it.

IrishCoffee17, 2021-04-23:
BBC, Capital London, Heart London are examples which should also broadcast pics.

srcejon, 2021-04-23:
I saw that from a search, but currently doesn't appear to be the case. Anyway, I've guessed what a fix might be. You should just need to type ninja to rebuild and the patch to the DAB library should be picked up automatically.

IrishCoffee17, 2021-04-23:
Your patch proofed to work! Listening different DAB+ programs since 5 minutes with no issues. Never before had come that far. Thanks!

srcejon, 2021-04-23:
Excellent. Have a look in your temp directory (possibly C:\Users\you\AppData\Local\Temp\), and you may see the images there.

IrishCoffee17, 2021-04-23:
Yes under exaxtly this path I find the pics (static station logos), but they seem to be inconsistently decoded. 2 examples given below
[image]
[image]

srcejon, 2021-04-23:
If you can send another sdriq file, I'll take a look.

IrishCoffee17, 2021-04-23:
Stupid question for the DAB demod - how would you change channel whilst listening so that also the Programs list updates? My workaround is to create one preset per desired channel and switch between presets. 

Listening DAB since 1 hour without the slightest problem. Much appreciated!

New sdriq recording (90 seconds) of another local MUX (Channel 5D) as .zip here:
Spectrum looks a bit strange which is normal for a SFN station.
Don't know which contained programs broadcast pics, but * 88.6 * does for sure, just tested. Also this pic appears puzzled when opened in the Temp folder.
[https://drive.google.com/file/d/1heNH9rfZV_y7NyVOG_qcMJ3TvYCiQ3Hs/view?usp=sharing](https://drive.google.com/file/d/1heNH9rfZV_y7NyVOG_qcMJ3TvYCiQ3Hs/view?usp=sharing)

srcejon, 2021-04-23:
Currently, the program list should contain all the programs for the tuned ensemble. When you switch to a new frequency, new programs will be added for that frequency, but as soon as you change the frequency, the program you are currently listening to will stop.

A feature that isn't yet implemented is for a channel scan - which goes through all of the channels and records a list of all programs there are and then saves it, so you can always see the complete list.

Thanks for the file, I'll take a look.

IrishCoffee17, 2021-04-23:
Aaaah the programs get appended to the list. All right, this is working. I had never scrolled the programs further down :)
Also you may want to consider my earlier [comment ](https://github.com/f4exb/sdrangel/issues/864#issuecomment-825036150) for the installation guide. It removes the need for an extra cmake install but unfortunately takes about 2 GB of diskspace.

IrishCoffee17, 2021-04-23:
Does not surprise me :)

IrishCoffee17, 2021-04-23:
Thanks for updating the install prerequsites. I see the 2 references to cmake are still in?

srcejon, 2021-04-23:
Ok - fixed that bug too - and when the above pull request is merged, you'll be able to see the images in the GUI.

Note, changes to the SDRangel source aren't picked up automatically when you build (like there are for the DAB library). So after the merge is complete, you'll need to do a 'git pull' to get it.

f4exb, 2021-04-23:
Tried broadcast images. Works great. Thanks!

IrishCoffee17, 2021-04-23:
I have a debug build ready now, but it has no support for a PlutoSDR device. This currently my only available device.

f4exb, 2021-04-23:
Yes you do need `libiio` for PlutoSDR support.

IrishCoffee17, 2021-04-23:
In the existing build directory I did a rerun of cmake without the `-DENABLE_IIO=OFF` followed by `ninja` and `ninja package`, but have still no PlutoSDR in the build.

f4exb, 2021-04-23:
That's `-DENABLE_IIO=ON` that you need

IrishCoffee17, 2021-04-23:
Yep thanks. Listening to DAB now but sorry, no MOT image on the Windows GUI. Where should that be?

f4exb, 2021-04-23:
Not all stations use MOT and it may take a little time before it comes up. When it does it appears in the "Current Program" section:
[image]

IrishCoffee17, 2021-04-23:
Aha maybe I was too impatient. On most programs it takes quite a while. Nice!
[image]

IrishCoffee17, 2021-04-23:
My presumably last question on this thread is, if there is any chance to get rid of the Windows build errors for rtl-sdr?

srcejon, 2021-04-23:
I don't know - I just did a clean build and it worked fine.

You might try doing the same - i.e. checking out all of the source from scratch somewhere else and see if the same problem reappears.

IrishCoffee17, 2021-04-24:
You did that in Windows or Linux? 
In Windows I started multiple times from scratch, even on two different machines. Build always stops at the same point as documented above. Reference is made to a non-existing directory. Error in build config?

srcejon, 2021-05-05:
Yep, on Windows. I've just built it again from scratch and it is OK.

I think it may be related to MFC not being installed as part of Visual Studio.  Could you please try adding that?

IrishCoffee17, 2021-05-05:
Confirmed. Adding the MFC option did the trick. Thanks.

closed 2021-05-09 by srcejon