f4exb/sdrangel #865: APT demod: processPixels takes too much time on the main thread

f4exb, 2021-04-22:
The call to method `processPixels` [here](https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demodapt/aptdemod.cpp#L174) (mostly what's behind) takes too much time and blocks the main thread for too long. This has quite undesirable side effects. The most visible is that the main spectrum is not moving smoothly anymore. Things gets worse as the image grows because the full image is processed at each new line so you have to wait a little bit to see the effect. 

More annoyingly if another plugin is present then its flow is disrupted which results for example in breaks in audio (for a plugin with audio) with messages showing that the sample FIFO is not processed fast enough.

The simplest fix is to move this process to a separate thread as I am not sure the process itself can be optimized by processing only the current line. There must be a good reason to reprocess the full image as I have noticed that it improves over time.

A fix is going to be posted to`master` shortly.

srcejon, 2021-04-22:
I was a little worried that might be too CPU intensive, but hadn't noticed a problem (although didn't look too hard), so didn't spend much time on it. 

Another thread sounds good, but I would add that it's not really necessary to call processPixels for every row, so if you wanted to actually reduce the amount of processing as well, you could do that easily by only applying the filtering every n rows that are received. In fact, in the aptdemod class, it really only needs to be called once after the entire image has been received, before writing to disk. It's just done every row so the result is immediately seen in the GUI as the image is being received. However, that continual processing could be moved to the GUI thread - and split in to the filters that work per row and those that work over multiple rows etc. There's probably also quite a bit of unnecessary copying that happens too.

f4exb, 2021-04-22:
I have experienced a bit with the separate thread and this is good enough to make things smooth. I had the feeling that some more optimization is possible but didn't want to take too much risks as a first step. I also found a bit strange that the whole image is processed at every new row and I take that you confirm this is not strictly necessary. Of course something may be not always optimal has to happen on the GUI so that the user thinks that the plugin is not stalled.

f4exb, 2021-04-23:
Just posted a new patch to do a full image processing only every 20 lines. To show some activity to the user the unprocessed line is sent for the other lines and displayed as is.

srcejon, 2021-04-25:
Quite possibly a problem that already existed before these changes - but after leaving it running overnight for a few passes, I noticed memory usage had gone from a couple of hundred MB to a couple of GB - so we may have a bit of a memory leak.

f4exb, 2021-04-25:
I have spotted two possible places and fixed them in e8dec5db8 not sure it is THE issue though,

f4exb, 2021-04-26:
Deployed in v6.10.2

closed 2021-04-26 by f4exb