f4exb/sdrangel #870: Access Violation whilst changing between Presets

IrishCoffee17, 2021-04-25:
Apologies, me again.
Thanks for fixing #868.
Windows 10, latest SDRangel patch level.
Did only observe when switching from any preset to a preset with DAB demod only. About once every 10 times.  

[image]

f4exb, 2021-04-25:
That's a bit too deep. Do you have a stack trace?

IrishCoffee17, 2021-04-25:
>	[Inline Frame] sdrbase.dll!Sample::operator=(const Sample &) Line 52	C++
 	[Inline Frame] sdrbase.dll!std::_Copy_unchecked(const Sample *) Line 4421	C++
 	[Inline Frame] sdrbase.dll!std::copy(std::_Vector_const_iterator<std::_Vector_val<std::_Simple_types<Sample>>>) Line 4457	C++
 	sdrbase.dll!SampleSinkFifo::write(std::_Vector_const_iterator<std::_Vector_val<std::_Simple_types<Sample>>> begin, std::_Vector_const_iterator<std::_Vector_val<std::_Simple_types<Sample>>> end) Line 186	C++
 	sdrbase.dll!DSPDeviceSourceEngine::work() Line 352	C++

IrishCoffee17, 2021-04-25:
More context:

DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=1: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2083336: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2083336: 8
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2083336: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2083336: 8
DevicePlutoSDRBox::set_params: set attribute xo_correction=39999750: 9
DevicePlutoSDRBox::set_params: set attribute out_altvoltage0_RX_LO_frequency=86815000: 9
DevicePlutoSDRBox::set_params: set attribute in_voltage_rf_bandwidth=1500000: 8
DevicePlutoSDRBox::set_params: set attribute in_voltage0_rf_port_select=A_BALANCED: 11
DevicePlutoSDRBox::set_params: set attribute in_voltage0_gain_control_mode=slow_attack: 12
DevicePlutoSDRBox::set_params: Unable to write channel attribute in_voltage0_hardwaregain=40: Unknown error (-95) 
DevicePlutoSDRBox::set_params: set attribute in_voltage_bb_dc_offset_tracking_en=1: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_rf_dc_offset_tracking_en=1: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_quadrature_tracking_en=1: 2
The thread 0x1bcc has exited with code 0 (0x0).
Exception thrown at 0x00007FFEE48BBDE4 (sdrbase.dll) in sdrangel.exe: 0xC0000005: Access violation writing location 0x0000019264E49060.

f4exb, 2021-04-26:
Unreproducible in Linux.

f4exb, 2021-04-26:
At:
```
sdrbase.dll!SampleSinkFifo::write(std::_Vector_const_iterator<std::_Vector_val<std::_Simple_types>> begin, std::_Vector_const_iterator<std::_Vector_val<std::_Simple_types>> end) Line 186 C++
```
which is here:
```
                len = std::min(remaining, m_size - m_tail);
                std::copy(begin, begin + len, m_data.begin() + m_tail); <===============
```
Is it possible for you to show the values of `remaining`, `m_size` and `m_tail` ?

f4exb, 2021-04-26:
Other FIFOs also deserve review of mutex lockers.

IrishCoffee17, 2021-04-26:
@f4exb good job!
Yes with pleasure, I will try to determine those three values later today.

f4exb, 2021-04-26:
This should work just judging by the counters. I have commited e3910c0aa with the mentioned mutex fixes but I doubt this will solve the issue.

IrishCoffee17, 2021-04-26:
Works fine for me. I tried hard to reproduce the bug but SDRangel keeps stable.

IrishCoffee17, 2021-04-27:
Tested fix on two Windows 10 machines and three devices (PlutoSDR on RX & TX, Soapy RTL-SDR, SDRplay RSP2) wihout any issues.
This does now appear to be the most robust Windows build of SDRangel ever. @f4exb big thanks!

closed 2021-04-27 by IrishCoffee17