f4exb/sdrangel #872: Feeding multiple traces to scope

srcejon, 2021-04-26:
Hi,

I'm trying to use the scope to help debug a demodulator. This is pretty good at displaying two signals from the demod, but it would be helpful if it could display more than 2 simultaneously (I've set up 15 different signals to probe, but not all need to be visible simultaneously). 

Currently, I have combo boxes that select the different signals to send to the scope on the real and imag parts of the input. I've tried embedding two scopes, for 4 signals, but the triggering isn't synchronised, and so they get out of sync (unless only start/stop acquisition is used)

It seems the scope can display more than 2 traces in the waveform display, but these only come from one IQ input, as far as I can see? Is there a way to support more input channels?

[image]

This is perhaps similar to #16

f4exb, 2021-04-26:
Yes similar but not exactly the same. #16 is for a channel analyzer that would accept a MIMO (MI that is) baseband thus being fed by multiple synchronous I/Q streams. Of course to make it really useful one would have to be able to combine traces from different inputs in the scope. So in fact this one would be a prerequisite for #16

f4exb, 2021-05-06:
The scope component being rather sensitive and brittle I'd rather make it a new multiscope component possibly carrying out some redesign. The scope is one of the oldest components and has probably accumulated a good amount of technical debt.

f4exb, 2021-05-14:
For the scope part this is to be covered by #898

f4exb, 2021-06-12:
See https://github.com/f4exb/sdrangel/releases/tag/v6.14.0 for details

srcejon, 2021-06-13:
Great, thanks. Will give it a try.

srcejon, 2021-06-22:
Just giving this a try. 

During initialisation, I call:

 m_scopeSink->setNbStreams(2);

Then in the channel in sampleToScope(), I have:

        Real r = std::real(sample) * SDR_RX_SCALEF;
        Real i = std::imag(sample) * SDR_RX_SCALEF;
        SampleVector m_sampleBuffer1;
        SampleVector m_sampleBuffer2;
        m_sampleBuffer1.push_back(Sample(r, i));                    
        m_sampleBuffer2.push_back(Sample(r, i));
        std::vector<SampleVector::const_iterator> vbegin;
        vbegin.push_back(m_sampleBuffer1.begin());
        vbegin.push_back(m_sampleBuffer2.begin());
        m_scopeSink->feed(vbegin, m_sampleBuffer1.end() - m_sampleBuffer1.begin());
        m_sampleBuffer1.clear();        
        m_sampleBuffer2.clear();  

But this results in a "vector subscript out of range" exception in:

[image]

with streamIndex=1

Is there something else I need to add / change?

f4exb, 2021-06-23:
This sbould work. From the screenshot it seems initialization is not complete when setNbStreams is invoked.

srcejon, 2021-06-24:
Thanks - it no longer crashes. 

However, I only see 0 in the "trace stream source" combo box. How should this be initialised? It would obviously be nice to initialise it with names corresponding to what each stream is.

Looking through glscopegui.cpp, I can't see anything that adds items to ui->traceStream, and Ui::GLScopeGUI* ui; is private, so I don't know how to access it from the ChannelGUI to add items. (E.g. ui->scopeGUI->ui->traceStream->addItem("Therehsold"); doesn't compiler)

f4exb, 2021-06-24:
There is very little room in the combo and certainly not enough for more that a few character. So having a relatively long word here like "Threshold" is a bad idea.
[image]

Yes Ui::GLScopeGUI* ui; is private, this is normal because it should be handled by GLScopeGUI exclusively. I think however that it is missing methods around ui->traceStream to effectively handle multiple traces.

f4exb, 2021-06-24:
Two new methods in `GLScopeGUI`:

  - `void setNbStreams(unsigned int nbStreams);`: sets the number of streams with default names (index numbers)
  - `void setStreams(const QStringList& streamNames);`: sets the stream names from a list of strings. Thus the number of strings sets the number of streams.

srcejon, 2021-07-08:
I'm not sure if you missed the comments in PR #939, but as you didn't reopen the bug, I thought I'd mention it again. It seems multiple streams do not work for the Radio Clock plugin. I just see 0s on the scope. (A single stream is fine).

srcejon, 2021-07-14:
I can now select the different streams for X trace, thanks.

However, it doesn't appear to work for Y. If I add a Y trace, I seem to get the same data on Y as X, regardless of the stream selection.

E.g. I have TH selected here - but the Y trace is showing the MagSq signal.

[image]

commit c4cef5fd (2021-06-23): ScopeVis: fixed setNbStreams method to allocate the memory lines on all streams. Fixes #872
commit bd04ee5a (2021-06-24): GLScopeGUI: effective implementation of stream input selection. Fixes #872
commit 95923e8b (2021-07-15): ScopeVis: never use projection cache. Fixes #872
closed 2021-07-15 by f4exb