f4exb/sdrangel #873: une erreur sur le SDRANGEL

gaspar1987, 2021-04-27:
[image]
Bonjour j'ai eu un problème avec une erreur sur le SDRANGEL Pouvez-vous me donner la solution pour le réparer?

f4exb, 2021-04-27:
Insufficient description.

closed 2021-04-27 by f4exb

gaspar1987, 2021-05-02:
Pourquoi ça ne fonctionne pas il faudrait que vous remettez moins de fichiers le fichier est 100 % lourd en sur les mémoire . Réduire démodulateur certains sont unity ne fonctionne pas même j'ai testé avec des antennes spécialisées et les amplificateurs et des réducteurs donc ça ne fonctionne pas c'est pour cela que je voudrais que vous faisais un logiciel version la nouvelle version meilleur que ça je ne vous critique pas mais j'espère une bonne qualité de sdranger que je puisse le faire marcher faire marcher avec la grande clé h Acker one il fonctionne bien quand on appelle avec mais le nouveau est bloqué parce que j'ai pris une version plus ancienne de 4.0 .c'est pour ça que je voudrais que le fichier soit moins lourd pour que on puisse utiliser en toute sécurité sans problème. Merci bonne journée

f4exb, 2021-05-02:
Qu'est-ce que c'est que ce galimatias incompréhensible? Ecoutez... si SDRangel ne vous convient pas tel qu'il est ou que vous n'y comprenez rien utilisez donc SDR# ou SDRconsole et fichez nous la paix. Merci!