f4exb/sdrangel #874: Cannot play DAB service with German Umlaut

andimik, 2021-04-27:
In Austria there is a DAB radio service called "jö.live" (with German Umlaut). 

This is the only service which sdrangel doesn't want to play after a click on it.

https://www.digitalbitrate.com/dtv.php?mux=5D&liste=1&live=164&lang=en

[image]

srcejon, 2021-04-27:
Is this on Windows or Linux? There are a few unicode issues to resolve on Windows, which may be related.

andimik, 2021-04-27:
It's Windows 10 64bit

andimik, 2021-04-27:
Well, I tried under Ubuntu 20.04 and this does **not** play as well. So it is not only Windows-related.

srcejon, 2021-04-27:
Ok, that should be fixed in the next release. Let me know if you see any other problems.

PR #875 (2021-04-27): Use UTF-8 rather than latin encoding for program names. 
Fixes #874.
commit b48a35d6 (2021-04-27): Use UTF-8 rather than latin encoding for program names. Fixes #874.
closed 2021-04-27 by andimik