f4exb/sdrangel #876: "illegal hardware instruction"

DC7IA, 2021-04-27:
On Ubuntu 20.04 LTS using version

`sdrangel                   5.9.0                       4      latest/stable  f4exb         -`

I get

`[1]    3064065 illegal hardware instruction (core dumped)  sdrangel`

I'm not sure what it means. I tried removing and reinstalling the snap, which did not work.

Please let me know what details I could provide, I don't really have an idea what that error means.

DC7IA, 2021-04-27:
Version 6.10.2 installed via .deb package works, but has the disadvantage of not being in a repository, thus not receiving automatic updates.

f4exb, 2021-04-28:
This version 5.9.0 .deb has not been built in Appveyor based on the appveyor.yml file based in this repository so is totally unsupported. I suspect it has been built on hardware that has AVX2 enabled without restricting to earlier SIMD support like SSE 4.2. However the .deb packages found in the releases section are forcibly limited to SSE 4.2

Secondly I then suspect that your hardware is compatible with SSE 4.2 but not with AVX2.

Nothing to be done.

DC7IA, 2021-04-28:
I don't need the latest, stable is good enough for me. :)

Is there a repository for the Debian package?

f4exb, 2021-04-28:
No, what is provided here in the releases section is one version for Windows 10 and one for Ubutnu 20.04. There might be other builds that I do not know of and maintained by their owners.

DC7IA, 2021-04-29:
I'd like to suggest making a repo or PPA. :)

closed 2021-04-29 by DC7IA

f4exb, 2021-04-30:
Sorry maintaining a PPA is too much hassle and not very exciting. I prefer to focus on development.